zmkfirmware/zmk #916: Poor connection between halves

wizar, 2021-08-17:
Setup:
- Shield: Jorne
- Board: nrfmicro v1.4

From time to time I'm experiencing some connection issues from right half to the left one. Usually, it happens if there is some obstacle between them (happens pretty frequently when I put halves between legs while lying down 😄 ). Is it expected and what I can do to fix it? FWIW I was trying to set `CONFIG_BT_CTLR_TX_PWR_PLUS_8=y` in the configuration, but it doesn't really help. Also, left half seems to have really good connection to the BT reciever, the issue is clearly in connection from right to the left

Ivan-Malinovski, 2021-08-21:
I don't know if this is at all useful to you, but I had an issue with connection, when I had put my batteries on **top** of the nice!nanos. When I moved the batteries underneath the PCB, connection dramatically improved.

wizar, 2021-08-22:
@Ivan-Malinovski Yeah, I tried to play around with as well - no luck. The weird thing is what master (left) works *really* well, even from another room. And slave (right) looses connection if there is any obstacle (or distance is more than 30cm) between it and master.

Is there any way to configure right half to become separate BT device and connect to PC instead of to the master half?

HoundThe, 2021-11-19:
@wizar Hi, have you figured out the reason or a solution? I have the exact same problem with my wireless Lily58 pro with nice!nano MCUs.

HoundThe, 2021-11-19:
Hmm so I've tried to take out my batteries from under the MCU and the connection improved by a lot. But I need to have the batteries placed somewhere and I feel like under the MCU is the only option.

wizar, 2021-11-19:
@HoundThe Not really, at the end of the day I soldered another pair of MCUs, but with external antena this time - works flawlessly :)

HoundThe, 2021-11-19:
Thanks for the response, I was actually successful with the enhanced BT power in the ZMK config. It made the distance good enough even with the batteries.

caksoylar, 2022-04-03:
[Current recommendation](https://zmk.dev/docs/troubleshooting/#connectivity-issues) is to increase TX power which also increases it for inter-half communication, closing.

closed 2022-04-03 by caksoylar

tomgroenwoldt, 2024-02-14:
This might be an old issue but setting `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y` in combination with `CONFIG_BT_CTLR_TX_PWR_PLUS_8=y` solved the poor connection between my two halves (nice!nanos).