zmkfirmware/zmk #927: 74HC595A Shift Registers Matrix Support ?

YCF, 2021-09-01:
87% keyboard have to use 10+  I/O pins to scan the row.

> Saving pins with a shift register:

> The astute among you probably noticed that I said a keyboard matrix requires a number of pins equal to the square root of the number of keys, but that I also said that my keyboard design only requires 11 pins. It should be 16, right? Nope, because we're using a 74HC595 shift register. This shift register lets us use only three of the Arduino's I/O pins to control up to eight output pins. Those three pins let us send a byte (eight bits) to the shift register, which sets its eight outputs pins to either HIGH or LOW. By using the shift register for the output row pins, we save 5 whole I/O pins!

> "So why not use a shift register for the input pins, too?" you ask. The simplest answer is that input requires a different kind of shift register and I didn't have that type on hand. But using a shift register for input also complicates how we read the columns and can cause issues with noise and "bouncing." Suffice it to say that it's a headache that I didn't need to take on in this case.

It is not to hard to add it to the QMK,but I'm not good at zmk's code.

more:
https://create.arduino.cc/projecthub/cameroncoward/64-key-prototyping-keyboard-matrix-for-arduino-4c9531

petejohanson, 2021-09-01:
As discussed on Discord (https://discord.com/channels/719497620560543766/719909884769992755/882621980497555516) this would ideally be done as a new GPIO controller driver, like is done for the Ferris in #642 that abstracts out the shift register, ending up with a DTS that looks like:

```

/dts-v1/

/ {
    shifter: 74hc595a@0 {
        compatible = "ti,xx74hc595a";

        a-gpios = <&gpio0 0 GPIO_ACTIVE_HIGH>;
        b-gpios = <&gpio0 0 GPIO_ACTIVE_HIGH>;
        c-gpios = <&gpio0 0 GPIO_ACTIVE_HIGH>;
    };

    kscan: kscan0 {
        compatible = "zmk,kscan-gpio-matrix";
        diode-direction = "col2row";

        col-gpios =
        <&shifter 0 GPIO_ACTIVE_HIGH>,
        <&shifter 1 GPIO_ACTIVE_HIGH>,
        <&shifter 2 GPIO_ACTIVE_HIGH>,
        <&shifter 3 GPIO_ACTIVE_HIGH>,
        <&shifter 4 GPIO_ACTIVE_HIGH>
        ;

        row-gpios = <&gpio0 0 GPIO_ACTIVE_HIGH GPIO_PULL_DOWN>;
    };
};
```

So the kscan driver doesn't need to be touched at all, and the shift register is merely a (limited to only outputs) GPIO port exposed for use by other GPIO code.

petejohanson, 2021-09-01:
Let's keep this open! Anyone wanting to implement the shift register driver can refer to my notes.

petejohanson, 2022-06-25:
This was actually added in #1325 Closing.

Note: Using these for output pins is preferred, since you can still use interrupts to detect initial key presses to avoid extra power draw from constant scanning.

closed 2022-06-25 by petejohanson