zmkfirmware/zmk #934: Keyboard hangs when OLED is put to sleep

dzervas, 2021-09-06:
I have a custom shield with a Particle Xenon board (nRF52840). Problem is when the OLED (SSD1306) should be put to sleep, the whole keyboard hangs.

Currently I'm on the commit https://github.com/zmkfirmware/zmk/commit/2a9ab828b53b70539621ccb4ec4fa0cd47afc959 and my zmk-config is https://github.com/dzervas/zmk-config/commit/f2845c9b2b0517d771e74f0d669a3f486400962a.

I get the following logs when I enable debug logging:
```
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
etup returned 0
[00:00:00.003,875] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.006,713] <err> bt_gatt: Failed to save Database Hash (err -2)
[00:00:00.006,744] <dbg> zmk.zmk_ble_ready: ready? 0
[00:00:00.006,744] <dbg> zmk.update_advertising: advertising from 0 to 2
[00:00:00.007,598] <dbg> zmk.zmk_sensors_init_item: Init MAIN_ENCODER at index 0 with sensor_number 0
[00:00:00.008,575] <inf> usb_hid: Device connected
[00:00:00.008,575] <dbg> zmk.zmk_usb_get_conn_state: state: 2
[00:00:00.008,605] <dbg> zmk.zmk_usb_get_conn_state: state: 2
[00:00:00.008,605] <dbg> zmk.get_selected_endpoint: Only USB is ready.
[00:00:00.008,636] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:00:00.008,636] <dbg> zmk.zmk_endpoints_send_report: usage page 0x0c
[00:00:00.008,666] <inf> zmk: Endpoint changed: 0
[00:00:00.008,758] <dbg> zmk.destination_connection: Address pointer 0x200089f2
[00:00:00.008,758] <wrn> zmk: Not sending, no active address for current profile
[00:00:00.008,758] <dbg> zmk.destination_connection: Address pointer 0x200089f2
[00:00:00.008,789] <wrn> zmk: Not sending, no active address for current profile
[00:00:00.008,850] <wrn> usb_cdc_acm: Device not configured or suspended, drop 1 bytes
[00:00:00.011,688] <inf> usb_cdc_acm: Device suspended
[00:00:00.011,688] <inf> usb_hid: Device suspended
[00:00:00.011,718] <dbg> zmk.zmk_usb_get_conn_state: state: 5
[00:00:00.011,749] <dbg> zmk.zmk_usb_get_conn_state: state: 5
[00:00:00.011,749] <dbg> zmk.get_selected_endpoint: Only USB is ready.
[00:00:00.012,756] <dbg> zmk.zmk_sensors_trigger_handler: sensor 0
[00:00:00.012,817] <dbg> zmk.zmk_keymap_sensor_triggered: layer: 0 sensor_number: 0, binding name: ENC_KEY_PRESS
[00:00:00.012,847] <dbg> zmk.on_sensor_binding_triggered: inc keycode 0x7004f dec keycode 0x70050
[00:00:00.012,847] <dbg> zmk.zmk_keymap_sensor_triggered: Behavior returned error: -35
[00:00:00.012,847] <dbg> zmk.zmk_event_manager_handle_from: Listener returned an error: -35
[00:00:00.017,547] <dbg> zmk.bvd_sample_fetch: ADC raw 4021 ~ 2945 mV => 4075 mV
[00:00:00.017,547] <dbg> zmk.bvd_sample_fetch: Percent: 84
[00:00:00.017,578] <dbg> zmk.zmk_battery_update: Setting BAS GATT battery level to 84.
[00:00:00.017,761] <inf> lvgl: lv_init ready
[00:00:00.018,127] <inf> lvgl: Object create ready
[00:00:00.018,402] <inf> lvgl: Object create ready
[00:00:00.018,707] <inf> lvgl: Object create ready
[00:00:00.018,859] <dbg> zmk.kscan_gpio_init_0: Configured pin 12 on GPIO_1 for input
[00:00:00.018,859] <dbg> zmk.kscan_gpio_init_0: Configured pin 11 on GPIO_1 for input
[00:00:00.018,890] <dbg> zmk.kscan_gpio_init_0: Configured pin 10 on GPIO_1 for input
[00:00:00.018,890] <dbg> zmk.kscan_gpio_init_0: Configured pin 2 on GPIO_1 for input
[00:00:00.018,920] <dbg> zmk.kscan_gpio_init_0: Configured pin 1 on GPIO_1 for input
[00:00:00.028,808] <inf> zmk: Welcome to ZMK!

[00:00:00.028,839] <dbg> zmk.kscan_gpio_configure_0: Configured GPIO 0
[00:00:00.039,001] <dbg> zmk.zmk_display_init:
[00:00:00.039,276] <inf> lvgl: Object create ready
[00:00:00.039,489] <inf> lvgl: Object create ready
[00:00:00.039,764] <inf> lvgl: label created
[00:00:00.040,496] <dbg> zmk.zmk_usb_get_conn_state: state: 5
[00:00:00.040,771] <dbg> zmk.zmk_widget_battery_status_obj: Label: 0x20006d44
[00:00:00.040,985] <inf> lvgl: Object create ready
[00:00:00.041,259] <inf> lvgl: label created
[00:00:00.042,480] <inf> lvgl: Object create ready
[00:00:00.042,724] <inf> lvgl: label created
[00:00:00.043,487] <dbg> zmk.set_layer_symbol: Layer changed to 0
[00:00:00.043,945] <dbg> zmk.zmk_display_init:
[00:00:00.089,050] <dbg> zmk.kscan_gpio_read: Sending event at 0,0 state on
[00:00:00.089,111] <dbg> zmk.zmk_kscan_process_msgq: Row: 0, col: 15, position: 15, pressed: true
[00:00:00.089,141] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 15, binding name: KEY_PRESS
[00:00:00.089,172] <dbg> zmk.on_keymap_binding_pressed: position 15 keycode 0x70028
[00:00:00.089,202] <dbg> zmk.hid_listener_keycode_pressed: usage_page 0x07 keycode 0x28 implicit_mods 0x00 explicit_mods 0x00
[00:00:00.089,202] <dbg> zmk.zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:00:00.089,233] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[... 45 lines cut ...]
[00:00:02.782,928] <dbg> zmk.zmk_kscan_process_msgq: Row: 1, col: 10, position: 26, pressed: false
[00:00:02.782,958] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 26, binding name: KEY_PRESS
[00:00:02.782,989] <dbg> zmk.on_keymap_binding_released: position 26 keycode 0x70013
[00:00:02.783,020] <dbg> zmk.hid_listener_keycode_released: usage_page 0x07 keycode 0x13 implicit_mods 0x00 explicit_mods 0x00
<some other keypresses>
[00:00:04.112,426] <dbg> zmk.zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:00:04.112,457] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:00:04.182,067] <dbg> zmk.kscan_gpio_read_0: Sending event at 2,13 state off
[00:00:04.182,342] <dbg> zmk.zmk_kscan_process_msgq: Row: 2, col: 13, position: 43, pressed: false
[00:00:04.182,373] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 43, binding name: KEY_PRESS
[00:00:04.182,403] <dbg> zmk.on_keymap_binding_released: position 43 keycode 0x70028
[00:00:04.182,434] <dbg> zmk.hid_listener_keycode_released: usage_page 0x07 keycode 0x28 implicit_mods 0x00 explicit_mods 0x00
[00:00:04.182,434] <dbg> zmk.zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:00:04.182,434] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:00:15.007,720] <dbg> zmk.zmk_usb_get_conn_state: state: 3
[00:00:15.008,392] <dbg> zmk.zmk_usb_get_conn_state: state: 3
[00:00:15.009,002] <dbg> zmk.zmk_usb_get_conn_state: state: 3
...
[00:00:15.463,470] <dbg> zmk.zmk_usb_get_conn_state: state: 3
[00:00:15.507,690] <err> i2c_nrfx_twi: Error on I2C line occurred for message 0
[00:00:15.511,535] <dbg> zmk.zmk_usb_get_conn_state: state: 3
[00:00:15.593,414] <dbg> zmk.zmk_usb_get_conn_state: state: 3
[00:00:15.594,116] <dbg> zmk.zmk_usb_get_conn_state: state: 3
...
```

lewurm, 2022-01-12:
Not sure if related, but I see this message as well after the sleep timeout of 900000ms (=15min) from 
https://github.com/zmkfirmware/zmk/blob/d29236141e1b425669e049bb4e5ae5b65903cd62/app/Kconfig#L345-L347

is hit:

```
[00:35:08.577,331] <dbg> zmk.zmk_usb_get_conn_state: state: 3
```

I have `ZMK_SLEEP` enabled: https://github.com/lewurm/zmk/blob/0292e29fe2e839574c9843dbae56cfb143765005/app/boards/shields/dactyl_iuab/Kconfig.defconfig#L25-L26

nmunnich, 2025-08-14:
Closing due to inactivity. If the issue hasn't been resolved yet, please turn to the ZMK discord for help.

closed 2025-08-14 by nmunnich