zmkfirmware/zmk #947: Apple "Globe key" modifier

dxmh, 2021-09-21:
Apple now use an additional modifier key: "Globe" (seen in bottom left of the keyboard below)

[image]

### Problem

It's not possible to send the Globe key with ZMK, but the Globe key is used heavily in recent versions of Apple operating systems. 

Here are some keyboard shortcuts in iPadOS 15 that use the Globe key:

> Go to Home Screen: Globe + H
> Show App Library: Shift + Globe + A
> Quick Note: Globe + Q
> Siri: Globe + S
> Control Center: Globe + C
> Notification Center: Globe + N
> App Switcher: Globe + Up Arrow
> Next App: Globe + Left Arrow
> Tile Window Left: Control (^) + Globe + Left Arrow
> Tile Window Right: Control (^) + Globe + Right Arrow
> Show Slide Over: Globe + Backslash \
> Move to Left Slide Over: Option (⌥) + Globe + Left Arrow
> Move to Right Slide Over: Option (⌥) + Globe + Right Arrow
> Source: https://www.imore.com/all-global-keyboard-shortcuts-ipad-ipados-15

I believe the Globe key is a repurpose/rebrand of Apple's own "Fn key" which has been around for a while. Apple's Fn key has its own keycode which is not usually available in custom keyboard firmware:

> Unlike most Fn keys, the one on Apple keyboards actually has its own keycode... sort of. It takes the place of the sixth keycode in a basic 6KRO HID report -- so an Apple keyboard is in fact only 5KRO. (Source: https://beta.docs.qmk.fm/faqs/faq_keymap#fn-key-on-macos)

### Potential workarounds

Previously, when this key was purely `Fn`, it was possible to work around the inability to send `Fn`+`8` for "Play/Pause" by instead sending the desired `C_PLAY_PAUSE`. However, these kinds of workarounds do not apply to Globe as there's no keycode for the desired result (e.g. "Tile Window Left").

Fortunately, Apple offers the ability to remap modifiers at the Operating System level on all their devices - it should be possible for ZMK to send Caps Lock and for the Apple OS to treat it as Globe. However, **this workaround is not currently possible for ZMK**, because accessing this settings pane with a ZMK keyboard causes the Apple device to lock up and reboot as per #378.

[image]

minjaep, 2021-09-25:
So it is not an 'issue' as Apple wants to make it a propriety key (maybe you need a license to implement it, as it seems like OS checks if the keyboard supports globe or not). I also have no problem using capslock instead of globe.

dxmh, 2021-11-13:
> I also have no problem using caps lock instead of globe.

I find the main problem with this approach is the inability to use capslock in [modifier functions](https://zmk.dev/docs/codes/modifiers/#modifier-functions) in the keymap. 

For example: `&kp LC(A)` in the keymap sends `LEFT_CONTROL`+`A`, but I don’t believe it’s possible to do the equivalent for `CAPSLOCK`+`A` (`GLOBE`+`A`).

daaaaaaaaaniel, 2022-03-16:
In QMK, there's a patch that enables the Apple fn [image] key. <https://gist.github.com/fauxpark/010dcf5d6377c3a71ac98ce37414c6c4>. How far off are we from being able to do the same sort of thing in ZMK?

petejohanson, 2022-03-16:
I am in full agreement with the QMK devs, we won't ever imitate a company's VID to enable a feature.

We would need to petition Apple to somehow add the VID we use to their allowed list, before bothering enhancing ZMK's HID layer to support this.

eytanhanig, 2022-09-23:
> we won't ever imitate a company's VID to enable a feature

They're over 15% of the market. Is there any legal reason we can't support this? It'd be unfortunate to punish users just because the OS vendor doesn't play well with others.

Hello9999901, 2022-09-23:
You can work around this by binding an unused key and remapping using `hidutil` on macOS. You can't do this on iPadOS, however.

petejohanson, 2022-09-24:
> > we won't ever imitate a company's VID to enable a feature
> 
> They're over 15% of the market. Is there any legal reason we can't support this? It'd be unfortunate to punish users just because the OS vendor doesn't play well with others.

Users are welcome to override the VID and PID for their local builds. The project itself won't ever do this by default, that would open us up to lawsuits from Apple or the USB-IF, in theory. 

The better way is to petition Apple to add our IDs, or change their software.

spencerthayer, 2023-02-05:
How does Keychron get away with this?

tinydogio-joshua, 2023-02-15:
> How does Keychron get away with this?

Seems to be fn2 on all my Keychron devices.

mtskf, 2023-03-24:
Not yet sorted?

or-agami, 2023-05-28:
> I find the main problem with this workaround is the inability to use capslock in [modifier functions](https://zmk.dev/docs/codes/modifiers/#modifier-functions) in the keymap.
> 
> For example: `&kp LC(A)` in the keymap sends `LEFT_CONTROL`+`A`, but I don’t believe it’s possible to do the equivalent for `CAPSLOCK`+`A` (`GLOBE`+`A`).

There is a workaround for using capslock in modifier functions?

marzelwidmer, 2023-07-30:
I facing the same issues...my workaround to use F1 etc. (my default are on my MacBooks F1 with FN key) to switch it on the OS

macOS Ventura
Choose Apple menu  > System Settings.
Click Keyboard in the sidebar.
Click the Keyboard Shortcuts button on the right.
Click Function Keys in the sidebar.
Turn on “Use F1, F2, etc. keys as standard function keys”.

LandonSchropp, 2023-08-04:
I'm using a workaround as well, but I wish we could support the globe character so that I could set consistent keyboard shortcuts between my laptop's keyboard and my Desktop keyboard.

Hello9999901, 2023-08-04:
Hey all, as Pete stated a year ago – "The project itself won't ever do this by default, that would open us up to lawsuits from Apple or the USB-IF, in theory."
Keychron shows up as an Apple keyboard (at least with `lsusb`), so I'm assuming they have some deal with Apple itself. Unfortunately, the tldr is that the Globe key will not be supported anytime soon (or ever). There are a variety of workarounds. My workaround uses `hidutil` and a little ways up the thread.
Just wanted to conclude some points of the discussion so far.

ldebritto, 2023-09-13:
> There is a workaround for using capslock in modifier functions?

One could use a macro for that:
```
  caps_a: caps_a {
            label = "CAPS+A";
            compatible = "zmk,behavior-macro";
            wait-ms = <0>;
            tap-ms = <0>;
            #binding-cells = <0>;
            bindings
                = <&macro_press &kp CAPS>
                , <&macro_tap &kp A>
                , <&macro_release &kp CAPS>
                ;
        };
```

eytanhanig, 2023-09-14:
> The better way is to petition Apple to add our IDs, or change their software.

@petejohanson What are the next steps for starting this petition?

petejohanson, 2023-09-14:
> > The better way is to petition Apple to add our IDs, or change their software.
> 
> @petejohanson What are the next steps for starting this petition?

1. Figure out how best to contact Apple,
1. Write up the petition, including a link to this issue,
1. Send it to Apple.

This issue is definitely not high on my list of priorities, so whoever is motivated to do the above is welcome to.

ReFil, 2023-09-20:
I've successfully made globe key work on a zmk powered board. It doesn't use the fn key approach in that QMK patch, and doesn't require any changing of the VID/PID, it is simply a rarely used consumer keycode.
[image]

orand, 2023-09-20:
Really? Are you the first person in the world to figure this out? If so, could you please elaborate? If not, where did you find how to do this?

ldebritto, 2023-09-20:
> I've successfully made globe key work on a zmk powered board. It doesn't use the fn key approach in that QMK patch, and doesn't require any changing of the VID/PID, it is simply a rarely used consumer keycode. [image]

So you’re using this key code? `C_AL_KEYBOARD_LAYOUT`? Or have you created a new one to post this Usage ID?

ReFil, 2023-09-20:
It's the 0x29D usage ID, I just added it to keys.h. it is specified in the apple accessory device specifications. There is no mention of the Fn key in said specification. It either uses the same code as the globe key or a proprietary method apple don't publish

arabshapt, 2023-09-20:
I just added that like and then used it in my config. the key did not register in karabiner eventviewer

ReFil, 2023-09-20:
Did you enable CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL=y in adv360_left_defconfig

ldebritto, 2023-09-21:
While it still does not register in Karabiner Event Viewer as `fn`, it does work with my keymap after also changing my `.conf` to include the `CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL=y` line.

Should anyone want to reproduce, follow these steps:

1. Edit your `.conf` file in the config folder to include the following line

```
CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL=y
```

2. Create a file in your config folder with the key definition. I've created the `apple_globe.h` file in my keymap folder with this single `#define` command:

```
/* Apple Globe key */
#define C_KEYBOARD_LAYOUT_SELECT (ZMK_HID_USAGE(HID_USAGE_CONSUMER, 0x029D))
#define GLOBE C_KEYBOARD_LAYOUT_SELECT
// As per PR 1938
```

3. Add the GLOBE key code to your keymap (mine is a &mt in the Z key)

```
&mt GLOBE Z
```

However it only seems to work if I have the keyboard connected via cable. When I use the keyboard connected via bluetooth, it just doesn't work.

So I guess there's still something else to change in keyboard settings. 

Still, many thanks for that, @ReFil ! I guess that we're getting closer to a solution.

arabshapt, 2023-09-21:
> Did you enable CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL=y in adv360_left_defconfig
I tried with CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL=y , still it did not work. It does not get registered... with wire or bluetooth

arabshapt, 2023-09-21:
Yes, it is enabled... I can try disabling it

arabshapt, 2023-09-21:
disabling it did not help

ReFil, 2023-09-21:
> While it still does not register in Karabiner Event Viewer as `fn`, it does work with my keymap after also changing my `.conf` to include the `CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL=y` line. [...]

If you use webhid explorer on a chromium based browser can you see the keycode working over bluetooth? https://nondebug.github.io/webhid-explorer/

Apologies for my relative lack of testing, I dont have a mac personally and have to borrow one for short periods of time to test it

arabshapt, 2023-09-21:
I hope we manage to make it work. Isn't globe or fn the same on mac? I would assume the difference is in hold and tap. 
How does adding apple_globe.h help at all? where do you reference that file? I added that line in ZMK repo(not zmk config)...
Do you have any idea whether one could add more modifiers that OS would understand? I am already using left and right modifiers as different modifiers. I think globe/fn could be used for than(in karabiner elements). But is there any way to also use something else as modifier?

ReFil, 2023-09-21:
I suspect this will not be compatible with NKRO in the same way that F13-F24 are not compatible with NKRO

caksoylar, 2023-09-22:
> However it only seems to work if I have the keyboard connected via cable. When I use the keyboard connected via bluetooth, it just doesn't work.

Note that changes like enabling NKRO or changing the consumer usage page size requires refreshing the HID report descriptor. For Windows, it seems to require the procedure described here: https://zmk.dev/docs/features/bluetooth#windows-connected-but-not-working. It is likely that macOS too will at least require a re-pair, if not a similar procedure.

ldebritto, 2023-09-24:
Following @caksoylar tip, I've fully reflashed my keyboard (both halves) and re-paired with my MacBook, now it works fine even via Bluetooth.

Karabiner Event Viewer does not recognize it as a keystroke, but other apps and the OS itself does. iPadOS as well.

Also, reporting back to @ReFil, this is the value from the HID Explorer on a Chromium Browser:
```
02 9D 02 00 00 00 00 00 00 00 00 00 00
```

arabshapt, 2023-09-24:
So did you have to turn off NKRO for it to work? Where do you use(import) apple_globe.h? Or is just creating it enough? Could you create a rule in Karabiner and check whether it can be used there even though karabinerviewer is not showing it?

arabshapt, 2023-09-24:
Does anybody know the technical reason why f13-24 do not work with NKRO anabled?

ldebritto, 2023-09-24:
@arabshapt , have a look at my repo. There you'll find all these answers. [ldebritto/zmk-config](https://github.com/ldebritto/zmk-config/tree/main/config)

ReFil, 2023-09-24:
Does it function as the Fn key as well as the globe key? I wasn't sure if apple handled the fn part of that key in a different way to the globe as it were as there's no mention of fn in the apple specification

ldebritto, 2023-09-24:
Actually, in Apple keyboards, `fn` and `globe` are physically the same. Recent keyboards use the globe in the keycap.

I’m not sure if this change is only cosmetical or if behind the key there is a different key code. For what is worth, in a world that many people would have mechanical keyboard, some of them would be very disappointed to realiza that their keyboard's fn key does not produce the behavior they expect when getting an Apple computer. Maybe that's the reason, but now I'm conjecturing.

In my testings, I have used a Magic Keyboard with the fn key and the my MacBook’s own keyboard (with the globe logo key). They map the same in Karabiner Event Viewer.

So I guess that even if they have different key codes, they map to the same outputs in the OS.

danieleavitabile, 2023-10-02:
I am relatively new in ZMK, and I would very much benefit from the Globe Key. Do I understand correctly from https://github.com/zmkfirmware/zmk/pull/1938 that we may have soon (or perhaps already have) a keycde now? If so, is it in a ZMK version that is still to be made public?

caksoylar, 2023-10-02:
@danieleavitabile Yes, if you build with an up-to-date version of ZMK (which is the default behavior if you followed the [user guide](https://zmk.dev/docs/user-setup)) then you should be able to use the `GLOBE` keycode that was added in that change.

caksoylar, 2023-10-02:
Closing since #1938 presumably fixes this, if anyone has issues with the keycode you can add a comment on this issue.

closed 2023-10-02 by caksoylar

danieleavitabile, 2023-10-02:
Thanks, if I understand correctly, I can just use the GLOBE code if I use GitHub actions for building the firmware

caksoylar, 2023-10-02:
Yes, unless you changed `config/west.yml` file, it will always build using the latest version of ZMK.

LeonB, 2023-10-02:
@ReFil very cool! Will this also be added to the Adv360 repo?

arabshapt, 2023-10-02:
On Ipad the globe gets registered, but on mac it does nothing for me... When I disable "modify events" in karabiner for my external keyboard(kinesis 360) then I see in unknown events in karabiner something(see image). The actual fn-key on my mac gets registered as 
  ```
  {
    "type": "up",
    "name": {"apple_vendor_top_case_key_code":"keyboard_fn"},
    "usagePage": "255 (0x00ff)",
    "usage": "3 (0x0003)",
    "misc": ""
  },
  ```
  using 0x0003 instead of 029d did not help...
[image]

ReFil, 2023-10-02:
What version of macOS are you using? It seems that your fn key is using the older custom usage page method

ReFil, 2023-10-02:
That's odd, is the hardware recent too? I was able to get it to bring up the emoji screen on an m1 macbook pro which was the same thing the globe key on the integrated keyboard did. If your hardware is older it might explain it

arabshapt, 2023-10-02:
m2 macbook. Does the following help?
[image]

arabshapt, 2023-10-02:
[image]

arabshapt, 2023-10-02:
> That's odd, is the hardware recent too? I was able to get it to bring up the emoji screen on an m1 macbook pro which was the same thing the globe key on the integrated keyboard did. If your hardware is older it might explain it

that makes me hopeful

ReFil, 2023-10-02:
Is it doing anything in the OS? as per [this comment](https://github.com/zmkfirmware/zmk/issues/947#issuecomment-1732450416) karabiner might not be able to detect it. Bear in mind it's possible the key is disabled in the  [MacOS keyboard settings](https://support.apple.com/my-mm/guide/mac-help/kbdm162/mac)

arabshapt, 2023-10-02:
my normal built in keyboard opens emoji panel, but kinesis does not open anything. On my second mac(macbook 16 2019) the fn keycode is the same 
```
{
  "type": "up",
  "name": {"apple_vendor_top_case_key_code":"keyboard_fn"},
  "usagePage": "255 (0x00ff)",
  "usage": "3 (0x0003)",
  "misc": ""
},
```

arabshapt, 2023-10-02:
is replacing 0x029d with 0x00ff enough? how can I add this myself to test whether it works?

ReFil, 2023-10-02:
Implementing the dedicated usage apple fn key is not trivial, are you trying this over USB or bluetooth?

arabshapt, 2023-10-02:
both. I could do the same thing you did in your PR or do you mean even more work than that?

ReFil, 2023-10-02:
It would be a lot more work to implement, where's your config repo? I can have a look to ensure the keyboard is correctly configured?

arabshapt, 2023-10-02:
here it is: https://github.com/arabshapt/Adv360-Pro-ZMK/tree/v3.0/config

ReFil, 2023-10-02:
at first glance this line shouldn't be there, if you're using full consumer usage it supercedes basic https://github.com/arabshapt/Adv360-Pro-ZMK/blob/1f5cc43ae89a16c9079f128913104f81f2520c9a/config/boards/arm/adv360/adv360_left_defconfig#L64C27-L64C27

arabshapt, 2023-10-02:
thank you, I commented it out)

ReFil, 2023-10-03:
It seems like the GLOBE keycode does not provide the full functionality of the globe/fn keys present on official apple keyboards, but does provide some of the functionality, With this in mind it might be worth reopening the issue or opening a new issue specifically about the Fn functionality

ickc, 2023-10-24:
I can confirm that the GLOBE keycode here is more limited than actual GLOBE key from Apple. I tried it on iPad Pro 12.9" 6th gen model with iPadOS 17.0.3, and it is working as expected (e.g. GLOBE key changes input sources.) But on the Macbook Pro 16" 2021 model with macOS Sonoma 14.0 (23A344), the GLOBE has no effects (e.g. it does not change input sources.)

Remapping in System Preferences > Keyboard > Keyboard Shortcuts... > Modifer Keys > Caps Lock to Globe gives full Globe/Fn functionality instead.

Also, in order to use the GLOBE keycode on Advantage 360, you need to follow https://github.com/zmkfirmware/zmk/issues/947#issuecomment-1743121985 and set `CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_BASIC=n`. I don't think it is documented yet in the Adv360 repo.

[3 more replies]