zmkfirmware/zmk #948: docs: Add a "feature" page on split boards/shields

caksoylar, 2021-09-21:
I noticed that there isn't a unified location on the docs on how split keyboards work with ZMK, while there are a few pieces of information in some pages such as the FAQ. Would it be helpful to have a "Split Keyboards" page, maybe under "Features" that details how they work?

A few things that come to my mind that we might document in such a page:
- Central/peripheral definitions clarified w.r.t. left/right halves
- Noting the peripheral only communicates with the central over BLE, then central to other devices through USB/BLE
- Two firmwares `_left` and `_right` are produced to be flashed to two halves
- Link to "New shield" page on developing new split shields
- "Dongle mode"-related information in the future can also be placed here and the page title could be changed to reflect the generalization to multiple peripherals

dxmh, 2021-09-23:
Related: https://github.com/zmkfirmware/zmk/pull/550/

dxmh, 2021-09-23:
It would also be good to cover how things like `&reset` behave on splits, etc.

Somewhat relevant PR:

* #547

caksoylar, 2021-09-23:
Added it to the list, thanks!

dxmh, 2022-02-12:
I think such a page would also be good place to document the frequently asked question of setting up left and right halves each as "central", along with some of the caveats such as no shared layer state, OS compatibility with cross-half modifiers (e.g. works on Linux, [requires additional software on macOS](https://www.dayid.org/comp/osx_dual-keyboard.html)), potential incompatibly with cross-half mod-taps, etc.

PR #2406 (2024-08-08): feat(docs): Add a split keyboards feature page
This is an explainer page to describe how split keyboards work in ZMK, with specific details on flashing, pairing, split communications and behaviors with locality. Fixes #948.
closed 2024-08-08 by caksoylar