zmkfirmware/zmk #1006: Momentary Layer not working when Toggle Layer is enabled

ialexey, 2021-10-31:
Momentary Layer modifiers `&mo` do nothing when I am on Toggle Layer `&tog`. It would be nice to have an ability to alternate layers while I am toggled.

Also it is only possible to leave the Toggle Layer by To Layer `&to`. `&tog` does nothing when toggled.

caksoylar, 2021-10-31:
This should work, and works fine on my keymap (through `&lt`s, but nevertheless). Is your `&mo` target layer "below" (i.e. defined earlier in the list) than your toggled layer? If so, that's how it is supposed to work because layers are triggered top to bottom and toggle layer would intercept momentary layer's key presses.

ialexey, 2021-11-01:
@caksoylar thanks for the fast response! The `&mo` target is defined earlier than the layer I am toggled to with `&tog`
The config is here: https://github.com/ialexey/zmk-config/blob/nice-nano/config/corne.keymap
I go to `default_alt_layer` from `lower_layer` with `&tog` and then try to `&mo` to `lower_layer`

ialexey, 2021-11-01:
Ok after moving my Toggle Layer above the layer I'm `&mo` to it worked! Thanks @caksoylar

closed 2021-11-01 by ialexey