zmkfirmware/zmk #1014: [Docs][Bug] copy custom selection of code removes line breaks in Firefox

auipga, 2021-11-03:
**Steps to reproduce:**
- go to https://zmk.dev/docs/behaviors/mod-morph (as on example)
- copy multiple lines with <kbd>Ctrl</kbd>+<kbd>C</kbd> or context menu:
  [image]

**Expected result on paste:**
```
        gresc: grave_escape {
            compatible = "zmk,behavior-mod-morph";
            label = "GRAVE_ESCAPE";
            #binding-cells = <0>;
            bindings = <&kp ESC>, <&kp GRAVE>;
            mods = <(MOD_LGUI|MOD_LSFT|MOD_RGUI|MOD_RSFT)>;
        };
```

**Actual result on paste:** 
```
        gresc: grave_escape {            compatible = "zmk,behavior-mod-morph";            label = "GRAVE_ESCAPE";            #binding-cells = <0>;            bindings = <&kp ESC>, <&kp GRAVE>;            mods = <(MOD_LGUI|MOD_LSFT|MOD_RGUI|MOD_RSFT)>;        };
```
---
Details: Firefox 94.0, Linux (Archlinux)
Cross-check with Vivaldi (based on Chromium): bug not present there.

joelspadin, 2021-11-03:
We need to pull in this fix: https://github.com/facebook/docusaurus/issues/5663

caksoylar, 2023-11-05:
This seems to be fixed now as I can't reproduce on Firefox 119.0 on Windows, possibly by the Docusaurus update in #1779. Please re-open if it is still an issue.

closed 2023-11-05 by caksoylar