zmkfirmware/zmk #1041: build fails on basic template that worked a few days ago

Incblob, 2021-12-01:
Hi, I'm uploading the basic template keymap, and the build fails with:

```
Run west build -s zmk/app -b nice_nano_v2 -- -DZMK_CONFIG=${GITHUB_WORKSPACE}/config -DSHIELD=a_dux_left 
-- west build: generating a build system
-- ZMK Config directory: /__w/zmk-config/zmk-config/config
-- Board: nice_nano_v2, /__w/zmk-config/zmk-config/zmk/app/boards/arm/nice_nano, a_dux_left, a_dux
-- ZMK Config Kconfig: /__w/zmk-config/zmk-config/config/a_dux.conf
-- Using keymap file: /__w/zmk-config/zmk-config/config/a_dux.keymap
-- Using keymap file: /__w/zmk-config/zmk-config/config/a_dux.keymap
Including boilerplate (Zephyr base): /__w/zmk-config/zmk-config/zephyr/cmake/app/boilerplate.cmake
-- Application: /__w/zmk-config/zmk-config/zmk/app
-- Zephyr version: 2.5.0 (/__w/zmk-config/zmk-config/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
-- git describe failed: fatal: No tags can describe '68c11efba6e1eb2d689b3039553868b1b1c57f94'.
Try --always, or create some tags.;
   BUILD_VERSION is left undefined
-- Board: nice_nano_v2, Shield(s): a_dux_left
-- Cache files will be written to: /__w/zmk-config/zmk-config/zephyr/.cache
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.12.4 (/opt/zephyr-sdk-0.12.4)
-- Found dtc: /opt/zephyr-sdk-0.12.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /__w/zmk-config/zmk-config/zmk/app/boards/arm/nice_nano/nice_nano_v2.dts
-- Found devicetree overlay: /__w/zmk-config/zmk-config/zmk/app/boards/shields/a_dux/a_dux_left.overlay
-- Found devicetree overlay: /__w/zmk-config/zmk-config/config/a_dux.keymap
Error: nice_nano_v2.dts.pre.tmp:735.297-298 syntax error
FATAL ERROR: Unable to parse input tree
CMake Error at /__w/zmk-config/zmk-config/zephyr/cmake/dts.cmake:205 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /__w/zmk-config/zmk-config/zephyr/cmake/app/boilerplate.cmake:535 (include)
  /__w/zmk-config/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /__w/zmk-config/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/__w/zmk-config/zmk-config/build -S/__w/zmk-config/zmk-config/zmk/app -GNinja -DBOARD=nice_nano_v2 -DZMK_CONFIG=/__w/zmk-config/zmk-config/config -DSHIELD=a_dux_left
Error: Process completed with exit code 1.

```
This worked last time I tried a few days ago,  and I don't know where to find the dts.pre.tmp file. 
Help?

petejohanson, 2021-12-01:
From our Discord bot command `!dtgithub`:

> GitHub build failing?
> * What is the error? It should look something like Error: nice_nano.dts.pre.tmp:760.12-13 syntax error. Remember the line number (760 in the example)
> * The next step in the GitHub build process shows the .dts.pre.tmp file. Find the line of the error. There is probably a typo there.

Have you customized your keymap?

Incblob, 2021-12-01:
I've fixed the keymap. But the nice!nanos aren't cooperating now :/ 
Thanks for the help

closed 2021-12-01 by Incblob

Add0z, 2023-01-30:
is this a keymap error? like something mistyped on the keymap or what?

caksoylar, 2023-01-30:
> is this a keymap error? like something mistyped on the keymap or what?

It is most likely with this kind of error message, please see the updated troubleshooting instructions at https://zmk.dev/docs/troubleshooting#west-build-errors. Also feel free to ask for help on [Discord](https://zmk.dev/community/discord/invite) help channels.