zmkfirmware/zmk #1043: Docs: Link to conditional layers page from layers behaviors page

petejohanson, 2021-12-02:
Currently, the [layer behaviors](https://zmk.dev/docs/behaviors/layers) page leaves the user guessing how to implement "trilayers".

It would be nice to note and link to the  [conditional layers](https://zmk.dev/docs/features/conditional-layers) page from there, to help users discover that feature.

In addition, the [keymaps](https://zmk.dev/docs/features/keymaps) page should probably link to the conditional layers feature as well.

cgoates, 2022-03-16:
Imma take a stab at this.  The desired link on the layer behaviors page is straightforward. What was the vision for a link from the keymaps page to the conditional layers page?  I don't see an obvious place to put it, since there don't seem to be links to the layer behaviors on the keymaps page either.

cgoates, 2022-04-07:
This is solved by #1218, which has been merged.

closed 2022-04-07 by caksoylar