zmkfirmware/zmk #1044: Docs: Behaviors "look up" page for learning about keymaps

petejohanson, 2021-12-02:
Currrently, if users are viewing a sample keymap file, and trying to understand all the behaviors they see there, they need to dig into our individual behavior docs to find each one.

It would be nice to have an aggregated index/lookup page, at the top of the "Behaviors" nested sidebar section, that gives has a table from "sample behavior" to the docs describing that behavior.

caksoylar, 2022-02-12:
Related: #525

adityarawat1223, 2023-09-30:
can I work on this? if yes can someone guide me on how should I proceed?

caksoylar, 2023-09-30:
You most certainly can, although it helps to be familiar with ZMK behaviors and the documentation structure. Pete's initial idea still sounds useful to me:

> It would be nice to have an aggregated index/lookup page, at the top of the "Behaviors" nested sidebar section, that gives has a table from "sample behavior" to the docs describing that behavior.

So you'd want to add a page like `docs/docs/behaviors/index.md`, add a link to it in `docs/sidebars.js`, then fill it in. Its contents can could be a table or markdown list of mappings from pre-defined behavior labels to links to the corresponding behavior docs page.
E.g.
```
- `&kp`: [Key press behavior](key-press.md)
- ...
```

It would of course be nice to have some preamble explaining the purpose of the page and give a bit of context.

petejohanson, 2023-09-30:
Yeah, that's exactly the high level jist of it. I don't know if the text description should be a bit longer for each one, or if we might add a small summary of the behavior, e.g.:

```
`&kp`: [Key Press Behavior](key-press.md) - Send HID keycodes to the connected host
```

huber-th, 2024-03-05:
This would be a very nice to have. @adityarawat1223 are you working on this? If not, I could take this one on.

Before I go through the whole list of behaviors, is this kind of what you had in mind @petejohanson ?

[image]

petejohanson, 2024-03-05:
> This would be a very nice to have. @adityarawat1223 are you working on this? If not, I could take this one on.
> 
> Before I go through the whole list of behaviors, is this kind of what you had in mind @petejohanson ?
> 
> [image]
> 

Exactly right! I'm not sure if it's useful to break the table into a few high level categories or not, if it ends up too long. I also had considered an "Examples" column showing how it looks to use it.

huber-th, 2024-03-05:
Great! I agree that it probably would make sense to identify some high level categories and break it up a bit to avoid a large single table. 

I'll see what might make sense to break it up and work on a proposal we can iterate on.

PR #2193 (2024-03-25): feat(docs): Add Behavior overview page
Resolves #1044 
Resolves #525 

@petejohanson this is an initial draft of the behavior overview page, structured with groupings I deemed sensible. While I'm not entirely satisfied with all the descriptions yet, I aimed to steer clear of excessive repetition from the detailed behavior pages to prevent duplication of content yet give a good idea to readers what each behavior is for.

I opted not to include an example column because the majority of examples would be quite basic, typically consisting of bindings and keycodes. Additionally, the provided links  [... cut]
closed 2024-03-25 by caksoylar