zmkfirmware/zmk #1045: Document `PAUSE_BREAK` and `SLCK` as macOS brightness controls

dxmh, 2021-12-03:
A discord user has reported the need to use the following keys to adjust display brightness on macOS:

* `PAUSE_BREAK`
* `SLCK`

From [the QMK docs](https://beta.docs.qmk.fm/using-qmk/simple-keycodes/keycodes#basic-keycodes-id-basic-keycodes), it seems that the keys `print screen`, `scroll lock` and `pause break` are "treated as F13-F15" on macOS. It also seems that in QMK the latter two keys are aliased to macOS specific brightness adjustments.

I'm unsure if these key codes are hardcoded to brightness adjustments in macOS, or its simply due to these default macOS keyboard shortcuts:

[image]

Either way, it would be good to confirm the F13-F15 behaviour and get something in the ZMK documentation to help steer users in the right direction. (Maybe even similar aliases for macOS specific brightness up/down?)

rmwphd, 2022-04-14:
A potentially useful note is that C_BRI_UP and C_BRI_DN on an encoder work great to adjust the brightness on a MacBook Pro in ZMK

robert-hill, 2023-05-29:
On all of my Macs, with up to date OS and up to date ZMK on my Kinesis Advantage 360 Pro I needed to use this workaround. `C_BRI_UP` and `C_BRI_DN` no longer work, though they did on my nice!60. I ended up settling on mapping F14/F15 instead because I only use Macs or an iPad and this appears to be the default mapping anyway. I do believe this to be some sort of regression but I am still working on gathering raw diagnostic logging between the two devices and configurations.