zmkfirmware/zmk #1053: Bug: wrong left brace keycode for german/ international layout

Incblob, 2021-12-11:
The LBRC key is mapped to 
```
/* Keyboard { (Left Brace) */
#define LEFT_BRACE                                                                                 \
    (LS(HID_USAGE(HID_USAGE_KEY, HID_USAGE_KEY_KEYBOARD_LEFT_BRACKET_AND_LEFT_BRACE)))
#define LBRC (LEFT_BRACE)
#define LCUR (LEFT_BRACE) // WARNING: DEPRECATED (DO NOT USE)
```
in  https://github.com/zmkfirmware/zmk/blob/main/app/include/dt-bindings/zmk/keys.h
which is defined using left shift. 
This keycode (and maybe more) is used in the german layout to send  the "ü" symbol. But since it's defined via a shifted input, it always sends the capitalized version "Ü". 
At least that's what seems to be the case, since I can't get the lowercase version. 

This should be changed to some non-shifted version to maintain compatibility with international layouts. 
I'm afraid I don't know which is the correct one to use, but clearly it must exist. 
ä,ß and ö work fine.

joelspadin, 2021-12-11:
That value is correct for sending a `{` character on a US layout. If you want the version of that which is not shifted, that is `[`, which is `LEFT_BRACKET`.

Incblob, 2021-12-11:
> That value is correct for sending a `{` character on a US layout. If you want the version of that which is not shifted, that is `[`, which is `LEFT_BRACKET`.

You are correct, I confused the two in the keymap I was referencing.

closed 2021-12-11 by Incblob