zmkfirmware/zmk #1067: Decide on social media policy for ZMK project

petejohanson, 2021-12-21:
For lack of a better spot to discuss it, starting this as an issue here.

## Context

Early on in the life of ZMK, I registered the https://twitter.com/ZMKFirmware twitter account as a way to share news about the project on Twitter. So far, only myself has ever had credentials/login to the account, and only I have posted to that account.

## The Impetus

Recently, with the announcement of the Kinesis Advantage 360 Pro, in my excitement, I tweeted (https://twitter.com/ZMKFirmware/status/1471201749692203023?s=20) a reaction to the announcement, and then posted a link to that tweet in the ZMK discord's `#announcements` channel.

Afterwards, a member of the community from a smaller, newer vendor (who I won't name unless they chose to name themselves) privately messaged me and noted that it could be seen as an endorsement of a commercial product. Upon having that pointed out to me, I agreed, edited my post in `#announcements` to remove the link, and sent a follow up there:

> A quick follow up: After posting the previous tweet, I did realize that doing so could be seen as endorsement of a specific commercial product, of a company that is new to ZMK, and certainly hasn't been actively involved in ZMK publicly. There are many other/smaller vendors who have been involved, contributing to ZMK, and helping the project move forward that I have not linked to in announcements  previously. I am excited for all of these developments, but to only share this one here is not honoring that. I'm removing the link to the tweet, and we can continue to discuss any and all ZMK projects in other channels. ❤️  to everyone involved in this great community.

There has been one other time (that I can easily recall) where I (as the ZMKFirmware account) retweeted a post from my personal twitter feed (@petejohanson): https://twitter.com/petejohanson/status/1402808092363038721?s=20 for the nice!60 release.

## The Discussion

I would like to solicit the community to help settle on a social media policy for the "official" channels (right now, Twitter and Discord) for how and when to link to projects, commercial or otherwise.

Some thoughts:

* Sharing exciting news for things happening in ZMK land is something I think benefits ZMK broadly.
* Sharing things that are commercial efforts creates questions about what should get shared, what shouldn't, how often, etc and raises questions about bias.
* Right now, I am the only one making these decisions, so how can we come up with a strategy to spread that decision making, without adding "too much process"? Do we have any volunteers who want to be the ZMK Social Media Manager?

Thanks again everyone!

Nicell, 2021-12-21:
I personally feel that it's fine to retweet and/or tweet about commercial products that are using ZMK. To me, it's clear that it isn't a direct endorsement of the product, but maybe we need to consider how to make that more clear. At the same time, it would be nice to give equal attention to projects that are large and small including OSH. Generally, I think the ZMK Twitter has been promoting more small projects than commercial products so far, which I think is good. It shows all the cool things people are tinkering with (corne blog post, artsey.io, split rp2040, etc) and shines a light on the community more than companies.

However, I think this issue comes more into play with how we deal with promoting various commercial products. How we accomplish equality in the promotion of commercial products is definitely not simple. I think it's a bit unreasonable to be aware of every commercial product and be tweeting about all of them equally. Maybe instead we can retweet/tweet about any commercial product we (Pete? Social media team?) find interesting and then upon request, we can also highlight other products if we missed them?

Metalish, 2021-12-21:
Hi. I can share thoughts only from the basic user perspective.

In my opinion, in order to have the same visibility for every project, I think a single post for each project would be appropriate, while at the same time keeping a list of all confirmed projects (as a link for example) with short descriptions and links to the creators and/or shops (if commercial).
I think this way, the scenario of "preferential endorsement" can be avoided as every board designer or vendor (if they wish ofc) would have a post with their project on ZMK's social media platforms.
Keeping a link with the list of all projects in each post, would also help in "proving" (for lack of a better word) that ZMK is not endorsing a specific product in particular.

If in the future, ZMK would like to promote or advertise a product, you can always add a big asterisk somewhere visible on the post, so its obvious to every one.

tabbeber, 2021-12-21:
As a complete novice of both coding, development, OS-projects, github and ZMK itself I'd say such tweets _might_ be seen as an endorsement. But, I think it could send an important signal that the ZMK project is interested in relationships and contributions from commercial partners in order to further the project. To me that feels like a good thing.

moergo-sc, 2021-12-21:
Something that I heard brought up about this particular episode is that 1) At the point of announcement, Kinesis had not appeared to be active in the community. 2) There are many other cool projects and products with participants very active and contributing to the ZMK community which had not been mentioned. 

Effectively ZMK will need to make a decision as to whether new large commercial vendors with no participation, or the community are more important to ZMK.

It is my opinion that for a product to be featured in ZMK official media, one of the requirement is at least some (token) level of participation in the ZMK community. 

ZMK will not grow if the big companies take and not give back. In fact in all likelihood such behaviour from big companies would retard the progress of the ZMK community, since the little vendors might need to keep their development under close-source to avoid the big companies free-loading (and then with their superior marketing muscle competing). 

Another area the ZMK needs to focus on is the brand use policy. At the moment there is no published ZMK brand use policy. As all logos etc has its own implicit copyright, using the ZMK logo without explicit permission can be construed as a copyright infringement (IANAL, but that's what I understand). For Glove80's use of ZMK logo on our website, I actually asked Pete for clearance. But apparently Kinesis didn't do that, and so potentially in a dubious position.

Nicell, 2021-12-21:
> It is my opinion that for a product to be featured in ZMK official media, one of the requirement is at least some (token) level of participation in the ZMK community.

This seems like a fair requirement for a product to be featured, although I'm not sure how we police participation effectively. Have a team member in the Discord helping people in the #help channel? Have x PRs created to improve ZMK in certain ways? Have x amount donated to ZMK on open collective?

> ZMK will not grow if the big companies take and not give back. In fact in all likelihood such behaviour from big companies would retard the progress of the ZMK community, since the little vendors [...]

For me personally, I'm always excited to see more people using ZMK on their keyboards. Whether it be a small or large project I'm always excited to see them using ZMK, and I never expect them to contribute something back *personally*. I made my work MIT licensed for a reason, and commercialization doesn't change that. That's how I feel currently, but I'd like to see how other contributors feel (since there are so many other contributors making ZMK happen), and I'm certain my view may change with time. I'd certainly prefer if those who use ZMK, large or small, contribute back directly, but I also believe them using ZMK attracts contributors indirectly. Many contributors haven't made hardware themselves. Because of this effect, I'd argue that they can help ZMK grow.

> Another area the ZMK needs to focus on is the brand use policy. At the moment there is no published ZMK brand use policy. As all logos etc has its own implicit copyright, using the ZMK logo without [...]

ZMK isn't a legal entity at the moment, so no trademarks/copyrights are explicitly owned by ZMK. [I created the current ZMK logo](https://discord.com/channels/719497620560543766/719544230497878116/720102684144828498) that's used, so I suppose I'm the copyright owner. I have no issue with commercial boards using it to advertise their use of ZMK personally. Maybe we should look into creating a 501(c)x organization around ZMK and work on brand use since ZMK use is definitely a feature worth advertising. Although, creating an organization requires a good amount of time, so probably better left for another discussion of its own.

moergo-sc, 2021-12-21:
> This seems like a fair requirement for a product to be featured, although I'm not sure how we police participation effectively. Have a team member in the Discord helping people in the #help channel? Have x PRs created to improve ZMK in certain ways? Have x amount donated to ZMK on open collective?

I guess that's for the community to decide, and IMO it should be really be based on common-sense. I am not suggesting that the bar should be high or unreasonable. I would also imagine that it would be fairer if the participation expected is somehow related to the vendor size. A one-person project is one thing, and a multi-million turnover company is quite another. 

However in this case, I think we have a pretty clear cut situation: From my conversations and from Discord, it was my understanding that Kinesis had never participated in ZMK Discord or otherwise.

petejohanson, 2021-12-30:
> Something that I heard brought up about this particular episode is that 1) At the point of announcement, Kinesis had not appeared to be active in the community. 2) There are many other cool projects [...]

Yes, this is definitely true, and the reason I retracted my post in our announcements channel.
> 
> Effectively ZMK will need to make a decision as to whether new large commercial vendors with no participation, or the community are more important to ZMK.

I don't think any one vendor is more important than any other. I can understand that my posting one specific vendor gave that impression, and I apologize for that.

I will post a generic follow up with some thoughts on how we might remove any "favoritism" here moving forward.

> 
> It is my opinion that for a product to be featured in ZMK official media, one of the requirement is at least some (token) level of participation in the ZMK community.

This I disagree with. I specifically chose the MIT license to attract *users* to the firmware (where "user" is both end user, and designers/companies that want to use it), with the idea that more users, of all kinds, will generally benefit the project, in various different ways.

Kinesis not approaching the project beforehand to contribute "Hey, look at these PRs" or "Hey, we donated $ to the project" isn't in my mind *exclusionary*.

> ZMK will not grow if the big companies take and not give back. In fact in all likelihood such behaviour from big companies would retard the progress of the ZMK community, since the little vendors [...]

The theory behind MIT, is that *yes* companies might benefit from having some "secret sauce", but they *also* benefit by the project they depend on surviving and progressing. In what *ways* these various vendor chose to engage with ZMK to do that I want to keep *open*, rather than dictate. I see no reason to take a hostile stance to *any* vendor, no matter what way they show up to the project, because I don't believe that long term benefits the ZMK project. 

> Another area the ZMK needs to focus on is the brand use policy. At the moment there is no published ZMK brand use policy. As all logos etc has its own implicit copyright, using the ZMK logo without [...]

Yes, there is a need for clarity here. I opened #1078 1078 specifically around that, in order to keep focus on this issue around the social media policy.

petejohanson, 2021-12-30:
> In my opinion, in order to have the same visibility for every project, I think a single post for each project would be appropriate, while at the same time keeping a list of all confirmed projects [...]

This is really what I am gravitating towards, as a generic principle.

In order to:

* Continue to promote ZMK generally, and
* To not curry any favoritism,

I would propose:

* The ZMKFirmware twitter bio have an addendum added, along the lines of "(re)tweets are not an endorsement or promotion of any project or product."
* A new Twitter List be created, owned by the `@ZMKFirmware` account, that is used to track retweeted/quote-tweeted accounts.
* The new policy for retweeting or quote-tweeting a given tweet is as follows:
  * The tweet must be something related to the ZMK project.
  * The tweet must adhere to the ZMK CoC standards: https://github.com/zmkfirmware/zmk/blob/main/CODE_OF_CONDUCT.md#our-standards
  * The tweet must either be:
    * From an account not currently in the maintained list of of previously quoted/retweeted accounts, OR
    * Contain significant news/information *about ZMK* not previously promoted. Examples:
      * Vendor reveals new keyboard X includes previously unsupported hardware, and they have created a new PR w/ a driver for it.
      * Vendor/user has submitted a PR to ZMK with a significant new contribution, technical, documentation, etc.
      * Vendor/user reveals their previously discussed product runs on hardware not previously used w/ ZMK, e.g. some fancy new RISV-C SoC.
      * This does *not* include events in the normal lifecycle of a product, such as a GB opening, stock becoming available of a previously mentioned project, etc.
 
This *does* give vendors an avenue to have a certain project discussed by our social media account more than once. But the way that would happen would be some *contribution to the project*, which is exactly the behavior we hope to encourage by folks that are using ZMK. So I don't really consider this a loophole, per se.

I do *not* think we need to have a committee or approval/sign-off process in place, *at this stage of the project*, but am open to re-evaluating that as we continue to grow.

Thoughts?

petejohanson, 2021-12-30:
As an additional follow up:

As the project creator and major contributor, I often tweet about stuff I'm working on for ZMK on my personal account (`@petejohanson`). I have *on occasion* retweeted those from the ZMK account. Are there any concerns w/ me doing that?

petejohanson, 2021-12-30:
Addition:

Any tweet that mentions the `@ZMKFirmware` account would have those criteria applied. Additionally, any tweet I discover, e.g. someone links it from our Discord channel, would also have the criteria applied.

Any account wishing to raise visibility for something could DM the Twitter account to request a certain tweet be evaluated for promotion. If they want to do so in advance, they are welcome to do so.

Part of the changes would be opening the Twitter account to accept message requests from anyone.

petejohanson, 2022-01-04:
Also, currently the "banner" for the Twitter account is a random pic of my Kyria which I got working w/ ZMK early on. I'd much rather we have a photo of a fully OSH board, so as not to have that be a semi-promotion as well.

[image]

Anyone have a high quality photo of an OSH keyboard running ZMK they'd like us to use instead? If we get multiple good submissions, we can cycle through them slowly.

Nicell, 2022-01-04:
Any of the pictures in [this album](https://imgur.com/gallery/E1qX7RT) are fair game. It's a Reviung41, which is OSH. Not the most fitting colors for ZMK though 😄

petejohanson, 2022-01-04:
> 
> However in this case, I think we have a pretty clear cut situation: From my conversations and from Discord, it was my understanding that Kinesis had never participated in ZMK Discord or otherwise.

Kinesis did not directly contact me or participate in our Discord server before the announcement. I was aware of the project by way of an intermediary who is working with them on their project, who informed me they were evaluating ZMK, asked some questions about various WIP features, etc, and has been working with them to implement items not found in ZMK.

dxmh, 2022-01-05:
> Anyone have a high quality photo of an OSH keyboard running ZMK they'd like us to use instead? If we get multiple good submissions, we can cycle through them slowly.

I really like the idea of cycling through multiple user submissions. There has been some great photos posted in `#show-and-tell` on the ZMK Discord server over time. It might be worth making an announcement on Discord asking for photo contributions?

The photo of my A Dux on [`app/boards/shields/a_dux/README.md`](https://github.com/zmkfirmware/zmk/blob/f692d64d0557f898ec08e26ec75320383d937337/app/boards/shields/a_dux/README.md) is available if you think it's suitable. (I feel like the obvious lack of TRRS connectors on that particular image somewhat emphasises the wireless nature of ZMK.)

petejohanson, 2022-01-06:
Switched to the A. Dux photo for now, I think a rotation might be a good call long term, yeah.

@Nicell Dig the photos, the only concern I have there is pics w/ RGB on probably set a different/tone/expectation for ZMK than maybe we want? "Look, you can have RGB and run wirelessly!" (Later: "This lasts an hour!")

krikun98, 2022-01-12:
https://github.com/yumagulovrn/dao-choc-ble
https://github.com/krikun98/Skean
Just a few suggestions.