zmkfirmware/zmk #1076: Feature Request: Better handling of keys with the same keycode but different modifiers

rayduck, 2021-12-27:
A common bigram in programming is `+=`, which may exist as two separate keys on a custom ZMK keyboard layout. Rolling the keys `+` and `=` will result in one of the key being ignored.

Underlying `+` and `=` is the same key, with the former under the shift modifier and latter under no modifiers. When a user types `+`, a keydown of `+`, i.e a keydown of `Shift` then `=`) is sent. If the user then presses the `=` key before `+` is released, it will be treated as a duplicate keydown event. The end result is `+` being typed, quite contrary to what the user have in mind (`+=`). Typing `=+` in the same manner also results in `=`, which I suppose reflects the sequence `=` keydown, `Shift` keydown, `=` keydown (ignored as `=` is already held down).

On a normal keyboard each key is, well, just one key. But the extensibility of custom keymaps means that the same keycode can exist in different forms (i.e. with different mods) on different key locations.

QMK has [fixed](https://github.com/qmk/qmk_firmware/pull/2710) this by forcing a key release when a new key press for the same keycode is registered.

innhyu, 2022-03-12:
Definitely something I hope is addressed. Doing any variable adds with += results in the = sign getting ignored!

pujvz, 2023-04-20:
I also hope that this is addressed at some point. I would like to help, but I do not know how.

aaronkollasch, 2023-05-17:
A solution that seems to work is to add a [key combo](https://zmk.dev/docs/features/combos) for the '+' and '=' keys, and bind it to a macro that taps '+' and '='. The combo can be specific to your symbol layer as well.

caksoylar, 2023-10-05:
Closing since #1828 solves this.

closed 2023-10-05 by caksoylar