zmkfirmware/zmk #1077: SYSTEM_POWER and SYSTEM_SLEEP keys don't work

exceptioncpp, 2021-12-28:
SYSTEM_POWER and SYSTEM_SLEEP keys don't work
Example:
`&kp SYSTEM_SLEEP` 
Expected behavior: put the PC to sleep
Actual behavior: nothing happen (USB or BT connection)

It works for the same keyboard but with Pro Micro + QMK and with others keyboards

Windows 11 21H2 22000.376
Sofle V2 + nice!nano v2
Latest ZMK

jcmkk3, 2021-12-28:
Have you tried the `C_SLEEP` or `K_SLEEP` codes from [this](https://zmk.dev/docs/codes/power) page? I don't even see the code that you're using on that page. Maybe it is one of the deprecated keycodes.

exceptioncpp, 2021-12-28:
Yes, I tried all *_SLEEP keycodes but only SYSTEM_SLEEP code is equal to other firmwares (including QMK that works).

Nicell, 2021-12-28:
This is most likely because we don't process the General Desktop usage page in general (which is what SYSTEM_SLEEP tries to use). See here: https://github.com/zmkfirmware/zmk/blob/main/app/src/hid_listener.c#L19
Currently, we only process keyboard and consumer pages, so `SYSTEM_SLEEP` won't work until we support the GD page.

urob, 2022-06-12:
I can confirm that none of the three options listed in the docs (`C_SLEEP`, `K_SLEEP`, `C_SLEEP_MODE`) works under Windows 11. If someone could confirm for Windows 10 as well, it would make sense to state so in the docs (currently all three codes are marked as "maybe working under Windows"). 

Here's a workaround using the new macro behavior that does work under Windows 10:
```
   macros {
        // Windows sleep key
        win_sleep: win_sleep {
            wait-ms = <50>;
            tap-ms = <5>;
            compatible = "zmk,behavior-macro";
            label = "WIN_SLEEP_KEY";
            #binding-cells = <0>;
            bindings = <&kp LG(X) &kp U &kp S>;
        };
    };
```
This defines a `&win_sleep` behavior that can be added to the keymap. Note that `wait-ms` must be sufficiently large for the macro to work. Setting it to 50ms works for me. I haven't tweaked it much, though, so one might be able to reduce it. If the macro only opens a menu without putting the computer to sleep, one should chose a larger value for `wait-ms`.

arshmalhotra, 2022-09-24:
Windows 11 definitely requires a higher value for `wait-ms`. I'd recommend around `<500>`.

Since the Kinesis Advantage 360 came out recently, this might be an issue for people using that keyboard and the "programming software" Kinesis provides. For those users, you need to pad the UI changes with `&none` because they don't provide `wait-ms` or `macro_wait_time`.

For the macro @urob mentions, something like this should work:
`bindings = <&kp LG(X)>, <&none>, <&none>, <&none>, <&none>, <&none>, <&kp U>, <&none>, <&none>, <&none>, <&kp S>;`

Personally, I'm using a different method:
```
macro_win_sleep: macro_win_sleep{
    compatible = "zmk,behavior-macro";
    label = "macro_win_sleep";
    #binding-cells = <0>;
    bindings = <&kp LG(D)>, <&none >, <&none >, <&none >, <&none >, <&none >, <&kp LA(F4)>, <&none >, <&none >, <&none >, <&kp UP>, <&kp ENTER>;
};
```

You can mess around with the number of `&none`s but the idea is to pad the next binding if there's a UI change.

drjayvee, 2024-01-31:
> Windows 11 definitely requires a higher value for wait-ms. I'd recommend around <500>. Using U and S doesn't work either so you'd have to have arrow key inputs.

On my (brand new) machine, 250ms or fewer is flaky. Timing aside, @urob's macro works fine for me, including `U S`.

jeremyong, 2024-05-04:
> This is most likely because we don't process the General Desktop usage page in general (which is what SYSTEM_SLEEP tries to use). See here: [...]

@Nicell Is there any issue tracking this underlying issue by any chance? Lack of system sleep is a pretty big bummer

caksoylar, 2024-05-06:
#1535 linked above would be the issue tracking it.

omzy, 2026-09-02:
I tried assigning Sleep (`KC_SLEP`) on my Keychron V6 Ultra keyboard, but it doesn't work, in both Windows 10 and 11:

[image]

(this is in the Keychron Launcher software)

Nothing happens when I press the button. Is there a fix for this yet?

caksoylar, 2026-09-02:
No, if there was this issue and #1535 would be updated.