zmkfirmware/zmk #1083: Multiple conditional layers makes the keyboard unresponsive

rpbaptist, 2022-01-03:
When defining multiple conditional layers, and then trying to access the conditional layer, makes the keyboard unresponsive.

This configuration works: https://github.com/rpbaptist/zmk-config-Corne-ish-Zen/blob/538172c624da1b9e2755b3b4a69d6c6b14b6d979/config/corne-ish_zen.keymap

When I define a second conditional layer, pressing both layer keys together immediately makes the keyboard unresponsive:

https://github.com/rpbaptist/zmk-config-Corne-ish-Zen/commit/c5ebe2234e17c709c6c28e8a5de4460ee8a688e3#diff-f207261a12211c44574c9c603ec810b9c3052c91f41b1d0a97011d0a353a520e

slashfoo, 2022-06-07:
I'm also encountering this issue, my ideal setup was going to be "enable layer 4 if any two of layers 1, 2, or 3 are active".

I implemented this as:

```
  conditional_layers {
    compatible = "zmk,conditional-layers";
    meta_one {
      if-layers = <1 2>;
      then-layer = <4>;
    };
    // meta_two {
    //   if-layers = <2 3>;
    //   then-layer = <4>;
    // };
    // meta_three {
    //   if-layers = <1 3>;
    //   then-layer = <4>;
    // };
  };
```

The snippet of code works as intended if the lines are left commented, but when uncommented I experience the issue described by @rpbaptist  when attempting to change layers.

petejohanson, 2022-06-07:
Can you try the linked PR I have open? If it works well, I will try to rebase/refresh and wrap the last requested changes.

slashfoo, 2022-06-08:
it works well in the sense of having the layers activate as expected. I ran into [a new bug (Bug#1339)](https://github.com/zmkfirmware/zmk/issues/1339) when testing the code.

closed 2022-06-08 by petejohanson