zmkfirmware/zmk #1086: planck_rev6 &bootloader doesn't work

orbekk, 2022-01-08:
The `&bootloader` key doesn't work with my planck_rev6. The keyboard appears to restart but boots back into ZMK when I press the `&bootloader` key.

It looks like [qmk_firmware](https://github.com/qmk/qmk_firmware/blob/755731f0ba98b9a0cf826cd3b6486dc9104a8824/platforms/chibios/bootloader.c#L78) writes a magic value that triggers the planck bootloader. Perhaps ZMK needs to do this.

urob, 2022-06-10:
+1 `&bootloader` doesn't work on my planck_rev6 either

urob, 2022-07-27:
Quick summary of @petejohanson's [comments on this](https://discord.com/channels/719497620560543766/719544230497878116/1001687124447334441) on discord: 

ZMK uses the Zephyr function `sys_reboot` (which in turn calls the device specific function `sys_arch_reboot`) for rebooting and resetting. The issue is that Zephyr does not currently have a `stm32`-specific implementation of `sys_arch_reboot`, causing it to fall back to a baseline version that does not support magic codes.

tl;dr: To fix the `&bootloader` for Planck (and other `stm32` devices), one would need to add a `stm32`-specific implementation of `sys_arch_reboot` to Zephyr itself.