zmkfirmware/zmk #1110: Wired split keyboard support

matrixik, 2022-01-31:
I saw in few places users requesting wired connection between split halves of some keyboards using ZMK firmware (like Advantage360 Pro or Glove80) and developers said that it is not supported by ZMK for now so this is tracking issue for this request.

Personally I would also prefer wired connection when I'm using my stationary computer so that I don't need to think about charging or constantly using additional USB  port (beside the one used for central role) when I have LEDs turned on all the time. And then using Bluetooth only on the go.

BeatScherrer, 2022-02-07:
I'm currently using the nice nano wired without any issues on my sofle v2. firmware on my repo but it's basically default generated. I don't even have batteries attached.

caksoylar, 2022-02-07:
ZMK doesn't support wired *communication* between two halves. The data connection between the two sides go through Bluetooth, even if you are *powering* each side with wires. Maybe this is not clear [in the FAQ](https://zmk.dev/docs/faq/#does-zmk-support-wired-split) item as well.

ai212983, 2022-02-08:
Technically it is possible (and was done by Glove80 guys) to treat both halves as separate keyboards. Then one can have fully wired connection via USB splitter for example.

cgoates, 2022-02-20:
This in progress PR is working toward wired communication between the two halves: https://github.com/zmkfirmware/zmk/pull/1117  It doesn't look like it's ready for testing at this point, but it is actively being developed.

MrMarble, 2022-02-20:
What about charging the board through the jack cable? That way with only one cable you can charge and use both half at the same time. 

I only have one type-c cable and when the keyboard runs out of battery I have to keep changing the cable to still be able to use it wile charging.

caksoylar, 2022-02-21:
> What about charging the board through the jack cable? That way with only one cable you can charge and use both half at the same time.
> 
> I only have one type-c cable and when the keyboard runs out of battery I have to keep changing the cable to still be able to use it wile charging.

This is not related to firmware but to the board you are using. As far as I know nice!nano doesn't support charging through TRRS while for instance [Mikoto](https://github.com/zhiayang/mikoto#ext_5v) does.

aureumleonis, 2022-03-04:
I'm really excited about this feature! Having one side die on me drives me crazy. As far as hardware goes, will it be possible to mix and match? E.g having a nice!nano on on the central side but a plain pro micro (and no battery) on the peripheral?

Percentnineteen, 2022-03-04:
> 

Pro micro doesn't speak zmk, so that won't work. I'm pretty sure you could get away with a port expander like the OG ergodox used to use right now (assuming you used a supported port expander)

petejohanson, 2022-03-04:
Any mixed solution needs to be mindful to not force polling the peripheral side, otherwise battery life will suffer drastically, FYI.

So things like four wire TRRS with an i2c expander would *work* today, the battery life wouldn't be good.

aureumleonis, 2022-03-05:
Interesting... do y'all have an intended hardware architecture for this use case? I'm planning a new build and want to make sure I have the correct pieces.

petejohanson, 2022-03-05:
> Interesting... do y'all have an intended hardware architecture for this use case? I'm planning a new build and want to make sure I have the correct pieces.

This will be more relevant for dual wired ZMK keyboards, e.g. running RPi pico/rp2040 on each half eventually.

0x64746b, 2022-06-07:
I would love to run ZMK on any board using Pro Micros so that I
* can experiment with how the mod tap options differ from what QMK does
* later just switch out the Pro Micros with a pair of n!n to make the whole thing wireless (and maybe even switch back, when I borrow the n!ns for the next project ;))

petejohanson, 2022-06-07:
> I would love to run ZMK on any board using Pro Micros so that I
> * can experiment with how the mod tap options differ from what QMK does
> * later just switch out the Pro Micros with a pair of n!n to [...]

Note: this is about wired split support for supported chipsets/architectures. ZMK will never support AVR/atmega based devices like the original pro-micro, because Zephyr doesn't support those 8-bit devices at all,not never will.

Davides98, 2023-10-14:
there are some news?

petejohanson, 2023-10-14:
> there are some news?

There is an updated PR https://github.com/zmkfirmware/zmk/pull/1954 which is a new effort to make this move forward. That's hopefully going to move this forward.

tcaxle, 2023-10-18:
I have just built a Sofle V1.1 using Nice!NanoV2s and it seems to work pretty seemlessley with split-wired. I have USB to the left shield and TRRS connecting left to right and keypresses from the right sheild are registered just fine on the connected computer.

My build is wired-only and does not have batteries. If you are looking to do a similar thing (want the functionality of ZMK but aren't bothered about the wireless bit) then I can confirm that it does work.

petejohanson, 2023-10-18:
> I have just built a Sofle V1.1 using Nice!NanoV2s and it seems to work pretty seemlessley with split-wired. I have USB to the left shield and TRRS connecting left to right and keypresses from the [...]

That is not using wired data transport, you're just powering the other side over the TRRS cable, which may work, but is not advised, due to risk of damage to the controller if done incorrectly. Please just power both sides with USB after removing the TRRS while all USB is removed from the keyboard first.

tcaxle, 2023-10-18:
> That is not using wired data transport, you're just powering the other side over the TRRS cable, which may work, but is not advised, due to risk of damage to the controller if done incorrectly. [...]

Interesting. What exactly is the risk?

Percentnineteen, 2023-10-18:
First, plugging or unplugging the cable under power causes a momentary
short from power to ground. The charging IC is not very tolerant of that
condition and frequently fries as a result.

Also, because the power path is undocumented there is no way to know if you
are risking damage to any of the components in that path.

tcaxle, 2023-10-18:
Reasonable concerns, thank you for explaining.

tcaxle, 2023-10-19:
> First, plugging or unplugging the cable under power causes a momentary short from power to ground.

How will these hardware issues be addressed when wired data transport is implemented? Surely the risk of the momentary short stays the same regardless of whether the TRRS cable is being used for power delivery or data transfer?

> Also, because the power path is undocumented there is no way to know if you are risking damage to any of the components in that path.

What do you mean by undocumented? For n!n-v2s at least, the full schematic is online: https://nicekeyboards.com/docs/nice-nano/pinout-schematic

Percentnineteen, 2023-10-19:
Wired does not necessarily mean TRRS. Wired transport will (hopefully)
include a massive disclaimer about this issue when it is officially
supported. TRRS should never have been used in this manner, because it
ALWAYS shorts adjacent pins on insertion or removal but the previous
designs were tolerant and less likely to fail.

Second, yes the top level design is documented but the actual electrical
path through them is not something documented in the documentation for
those components from their manufacturer. Think of them as being off-label
use for pharmaceutical drugs. Do they do something potentially useful?
Yeah. Have they been studied for harmful side effects? No.

tcaxle, 2023-10-19:
> Wired does not necessarily mean TRRS. Wired transport will (hopefully) include a massive disclaimer about this issue when it is officially supported. TRRS should never have been used in this manner, [...]

Thank you for clarifying.

> Second, yes the top level design is documented but the actual electrical path through them is not something documented in the documentation for those components from their manufacturer. Think of [...]

What do you mean by this? The datasheets for all the parts are readily available online, and the voltage regulator and battery charging IC both have internal block diagrams. Or do you mean that the parts fitted on the controllers won't necessarily be on-brand, i.e. they are using grey-market imitation parts in production?

Putting aside the short-circuit risk (which I agree with you about) the major issue with powering the peripheral controller via the TRRS connection seems to be that it will be running at 3.3V instead of 5V, but this seems to be within spec from a (brief) look at the datasheet for the MCU. Assuming they are on-brand parts, the voltage reg and the charging IC both have internal diodes that allow back-flow of current and all the necessary protection circuitry to deal with being used in reverse.

My proffessional opinion is that, as long as you take precautions not to (un)plug the TRRS jacks while the keyb is powered up, there is little risk of damaging anything. However, I would be hesitant to connect batteries to the peripheral controller without further investigation and I would avoid connecting anything to the USB port of the peripheral as it will not have the expected 5V supply.

The application is hacky and definitely not an intended use-case, but I would happily bet the cost of a Nice!Nano V2 that it works just fine.

From the schematics for the nice!nano V2, and the Sofle V1 (my application) and information from the relevant datasheets, you can summarise the power path as in the attached diagram. The salient point is that the central MCU is powered from the 5V supply taken before the voltage reg, but the peripheral one is powered from the 3.3V supply taken from after the central voltage reg (and back-flowing through the peripheral voltage reg)

[image]

Percentnineteen, 2023-10-19:
The chips are not off-brand, they are just being used in a manner that is
not described in the datasheets. As you say, the issue is that the
peripheral is being powered backwards through the charging chip and that
backwards path is not described anywhere in the datasheet for the part.
Things like current limits, voltage drops, etc are not known quantities in
this mode of operation.

I agree with you that it is almost certainly fine (for at least some
cases). In addition to the concerns you raised about plugging the
peripheral in to USB or hooking up a battery, things like current draw (via
LEDs, OLEDs, etc) may steer into more uncertain territory.

However, given the fact that so many people seem to fry the charging ICs
and the fact that it is relatively benign advise to just plug the
peripheral controller in via USB C, I do not believe this practice will
ever become anything more than discouraged.

From my perspective (assuming no batteries), doing the wrong thing requires
one extra wire (TRRS) while doing the right thing requires one extra wire
(USB C). But doing the right thing carries fewer caveats and restrictions.

tcaxle, 2023-10-19:
> The chips are not off-brand, they are just being used in a manner that is not described in the datasheets. As you say, the issue is that the peripheral is being powered backwards through the [...]

> I agree with you that it is almost certainly fine (for at least some cases). In addition to the concerns you raised about plugging the peripheral in to USB or hooking up a battery, things like current draw (via LEDs, OLEDs, etc) may steer into more uncertain territory.

True - my next port of call is to take some measurements of my keyboard running in this setup so see exactly what voltages are where and to check the stability of the rails when the keyboard is in use. Still waiting on my switches to arrive though so this will probably wait a week or so.

> However, given the fact that so many people seem to fry the charging ICs and the fact that it is relatively benign advise to just plug the peripheral controller in via USB C, I do not believe this practice will ever become anything more than discouraged.

I agree: it's not something I would advise someone to do without a suitable disclaimer, and if you are going to offer any advice then advice that will definitely not cause issues is preferable to advice that only hopefully won't cause issues...

> From my perspective (assuming no batteries), doing the wrong thing requires one extra wire (TRRS) while doing the right thing requires one extra wire (USB C). But doing the right thing carries fewer caveats and restrictions.

This is true, but the latter also requires an additional USB port, which I don't have spare. Someone above mentioned using a USB splitter, and that is probably the best compromise in this case, provided I can find one that has power and data on one fork and just power on the other.

Thanks for taking the time to talk this through.

Fethbita, 2024-01-21:
One comment from me about implementing this, our corporate policy states that we can not use Bluetooth keyboards. I would like to use split keyboards at work that have ZMK firmware, but as it currently stands, I can not.

sontags, 2024-08-12:
Just to make them visible, there are two (more) PRs likely related to this issue: #2080 and #2086.

Wired split was also teased in October 2023 (https://zmk.dev/blog/2023/10/05/zmk-sotf-6#coming-soon) by @caksoylar, I hope this is making some progress soon...

Nick-Munnich, 2024-08-12:
> Just to make them visible, there are two (more) PRs likely related to this issue: #2080 and #2086.
> 
> Wired split was also teased in October 2023 (https://zmk.dev/blog/2023/10/05/zmk-sotf-6#coming-soon) by @caksoylar, I hope this is making some progress soon...

There is also  #1954. That said, #2086 currently seems like the most promising PR to me at least. All of the reports of people testing said branch have been great, and the issues with the branch seem more to be needing additional features than functional issues.

NightWatchman, 2024-09-29:
I've got a lilly58 I built with two ProMicro NRF52840 controllers in a wired configuration with no batteries.

Not sure if anybody else has had this experience of having a working wired configuration, but I've been able to use both halves by plugging my USB-C into the left half, and using a TRRS cable to the right half, as long as I have `#include <dt-bindings/zmk/ext_power.h>` in the keymap. Both halves worked fine for months with no issues.

Yesterday though, the right-half suddenly just stopped working unless I plug a USB cable into it for power. I haven't been able to figure out why yet.

sontags, 2024-10-05:
@petejohanson just me being curios, not wanting to put pressure on this; there is a PR #2086 open for quite a while now and I wonder if you will be able to review the thing some time soon? As far as I can tell people seem do be happy with the implementation, and it appears to be a nice addition to the project... Thanks.

ulfbert-san, 2024-10-05:
> I've got a lilly58 I built with two NRF52840 controllers in a wired configuration with no batteries.
> 
> Not sure if anybody else has had this experience of having a working wired configuration, but I've been able to use both halves by plugging my USB-C into the left half, and using a TRRS cable to the right half, as long as I have `#include <dt-bindings/zmk/ext_power.h>` in the keymap. Both halves worked fine for months with no issues.
> 
> Yesterday though, the right-half suddenly just stopped working unless I plug a USB cable into it for power. I haven't been able to figure out why yet. My controllers are from aliexpress, and probably knock-offs so it's entirely possible that one of them just crapped out.

Did u find the solution? I've have the same problem. My right side isnt working.

NightWatchman, 2024-10-05:
The best suggestion I have at the moment is to ensure that you are including the `ext_power.h` headers when building your firmware. If you are using https://github.com/nickcoutsos/keymap-editor, then you must ensure one of your keys is set to one of the external power operations like `EP_TOG` or the header will be automatically dropped. You also must have a TRRS cable or some other means of power delivery from the plugged in half to the other half.

I'm waiting for replacement controllers to arrive. In my case it suddenly stopped working with no config changes, so my best guess is some sort of hardware issue.

Nick-Munnich, 2024-10-05:
For the legit nice nanos at least, the VCC pin has undefined behavior when used as a power input path. It stands to reason that it will degrade significantly quicker when used in that manner. I can easily imagine the same being the case for the knockoffs.

NightWatchman, 2024-10-05:
That's a good point. I hadn't thought about that once my keyboard was up and running.

Percentnineteen, 2024-10-05:
You are using the nice!nanos in an unsupported and undefined manner, so it
is likely that they just gave up the ghost. You are effectively powering
them by shoving power into a port that is intended to be output only.

vn971, 2025-01-02:
> Technically it is possible (and was done by Glove80 guys) to treat both halves as separate keyboards. Then one can have fully wired connection via USB splitter for example.

@ai212983 do you happen to have a link or any reference about that?

Dessix, 2025-01-03:
Under that scenario, it still requires synchronizing the modifier keys, because most people don't press right-shift for the right side's capital letters. If I recall, the Glove80 (Which is what I'm following this issue for) is simply doing this over Bluetooth, even when fully wired to the machine on both sides, something I'm looking to avoid.

caksoylar, 2025-01-03:
For Windows and Linux, modifier state is global so they can be activated by any keyboard. I think macOS handles them per-keyboard, which would be an issue if the two halves are separate keyboard devices.

vn971, 2025-01-03:
I think there are people (such as myself) that do the "programmable keyboard" part at software level. This way, the only thing that we need from the keyboard halves is to send a unique scancode whenever a key is pressed.

(For anyone interested, there are tools such as `kanata` that allow you configuring your keyboard quite extensively, including layers and tap-holds. I'd propose not to drift into discussing them though, and focus on whether a "wired" ZMK is possible, be it wired communication between the halves or sending scancodes from both halves to the main computer over the wire.)

ai212983, 2025-01-08:
> > Technically it is possible (and was done by Glove80 guys) to treat both halves as separate keyboards. Then one can have fully wired connection via USB splitter for example.
> 
> @ai212983 do you happen to have a link or any reference about that?

They have it in their [FAQ](https://www.moergo.com/pages/faqs), also see their [quick test](https://docs.moergo.com/glove80-user-guide/a-quick-test-drive-of-glove80/) instructions and there's more in the User Guide about [USB output mode](https://docs.moergo.com/glove80-user-guide/typing-with-glove80/#indicators). 

Please note, halves are still communicating with each other via Bluetooth.

vn971, 2025-01-08:
I was able to make a fully wired build of the Glove80. (Each half is an independent USB keyboard.)

If anyone is interested in that in the future, here are the changes that I've made: https://github.com/vn971/zmk/compare/upstream...vn971:zmk:main

You then need a small keymap to work with. Like this one: https://github.com/vn971/zmk-config/blob/master/config/glove80.keymap

If you want to try this build out, here's the firmware that you can flash on your Glove80 (it's tested to be working): https://github.com/vn971/zmk-config/actions/runs/12671985945/artifacts/2401620965  (the last downloadable artifact in https://github.com/vn971/zmk-config/actions/runs/12671985945)

Note that this is "cheating", as most of the QMK/ZMK-like features stop working this way. This is because each half is now essentially a USB keyboard. And each half just sends their unique scancodes and keypresses. To bring back all the layers, tap-dances and other smart functionality, you will need a software tool like `kanata`. to bring back layers, tap-dances and other smart functionality. Or, if this issue gets solved, then you'll have native wired support out-of-the-box.

P.S. Because the keyboard doesn't have any bluetooth or smart features this way, the firmware gets very small, too. It's `66.7 KB` for both halves together, or 8.5% of the original firmware size. Not really important for most use cases, but nice to have some additional affirmation that the bluetooth functionality is gone.

jvsqzj, 2025-02-02:
Greetings to all. I'm new to ZMK, and I'm starting to learn about all this for a very specific project I have.
I own a wired [DygmaRaise](https://www.theverge.com/22951481/dygma-raise-review-ergonomic-keyboard-review) split keyboard. It works with 3 USB cables, one for each half to connect to its SAMD21 microcontroller and a third cable going from the SAMD21 to a computer.

I already bought one nice!nanoV2, a battery and a spiraled slinky USB type C cable. I've decided to take on the endeavor of soldering another USB C port to the nice!nano and connect both halves of the Raise using I2C to a single nice!nano and have the full keyboard connect to a PC via bluetooth. I want to abstract this system as a single board. A new fully wireless version of this keyboard is being sold by Dygma, but I still want to do this project instead of upgrading.

I'm wondering if this idea still falls under the scope of this issue's concept of "wired split keyboard". Having both halves be connected by a single cable but still using bluetooth for the full keyboard sounds somewhat like a hybrid split keyboard (not fully wireless but still not physically connected to a PC).

petejohanson, 2025-02-02:
> Greetings to all. I'm new to ZMK, and I'm starting to learn about all this for a very specific project I have.
> I own a wired [...]

That's not *exactly* in scope for this, since we are focused on UART based wired split to start, but this certainly includes refactoring to make other split protocols easier to implement.

Silverdev2482, 2025-02-13:
Could an optional wired mode be implemented? So by default the parts would communicate wireless, but when it was detected that they were connected via wire, or manually triggered if monitoring for it was to resource intensive. That way you wouldn't have the hassle of the cord in between the two when you don't care about latency, but it is available if you do.

Nick-Munnich, 2025-02-13:
> Could an optional wired mode be implemented? So by default the parts would communicate wireless, but when it was detected that they were connected via wire, or manually triggered if monitoring for [...]

I believe that is planned, with the help of an extra pin used to detect the wired connection.

Silverdev2482, 2025-02-15:
I don't know too much about hardware stuff, why would an extra pin be necessary? Can't they just tell there is a voltage on the receive pin? If you were to wire it up so they can charge batteries off each other it would be a bit annoying to go from 4 wires to 5.

Silverdev2482, 2025-02-20:
Would these be the UART interrupt pins or a different dedicated pin?

nmunnich, 2025-03-31:
Closed by #2766

closed 2025-03-31 by nmunnich