zmkfirmware/zmk #1141: gcc-arm-embedded is not available anymore via brew

devenv, 2022-02-21:
Documentation says:
https://zmk.dev/docs/development/setup
`brew install --cask gcc-arm-embedded`

which was removed.

`brew install armmbed/formulae/arm-none-eabi-gcc` does exist, not sure if the same thing or not.

halcyonCorsair, 2022-02-22:
Are you  sure it was removed?
https://github.com/Homebrew/homebrew-cask/blob/master/Casks/gcc-arm-embedded.rb

devenv, 2022-02-22:
Hmm... I looked at some old post apparently, looks like it does exist.
Following the docs though gives me:
```
brew install --cask gcc-arm-embedded
Error: Cask 'gcc-arm-embedded' is unavailable: No Cask with this name exists.
```
Not sure if something is messed up in my setup, broke in Monteray, or some pre-requisite is missing from the docs 🤔