zmkfirmware/zmk #1144: Shield revision keymap file in ZMK_CONFIG ignored

puradox, 2022-02-22:
When executing the setup script, `Kyria Rev2` is shown as a keyboard option. After going through the setup script, `kyria_rev2.conf` and `kyria_rev2.keymap` files are created in `${ZMK_CONFIG}/config`

However, when building the separate `_left` and `_right` sides of the board, the created keymap file is not being used.

```
west build -p -d build/left  -b nice_nano_v2 -- -DZMK_CONFIG="/home/sam/Dev/zmk-config/config" -DSHIELD=kyria_rev2_left
west build -p -d build/right -b nice_nano_v2 -- -DZMK_CONFIG="/home/sam/Dev/zmk-config/config" -DSHIELD=kyria_rev2_right
```

For example,
```
$ west build -p -d build/left -b nice_nano_v2 -- -DZMK_CONFIG="/home/sam/Dev/zmk-config/config" -DSHIELD=kyria_rev2_left
-- west build: making build dir /home/sam/Dev/zmk/app/build/left pristine
-- west build: generating a build system
-- ZMK Config directory: /home/sam/Dev/zmk-config/config
-- Board: nice_nano_v2, /home/sam/Dev/zmk/app/boards/arm/nice_nano, kyria_rev2_left, kyria
-- Using keymap file: /home/sam/Dev/zmk/app/boards/shields/kyria/kyria.keymap
```

The expected behavior is to load `${ZMK_CONFIG}/config/${SHIELD_WITHOUT_LEFT_OR_RIGHT}.keymap`.

Marco-Christiani, 2022-02-27:
Is there a recommended temporary fix in the meantime?

puradox, 2022-02-28:
Renaming `kyria_rev2.keymap` to `kyria.keymap` will fix the issue. The same applies to the `.conf` file.

chrisgalvan, 2022-05-12:
Thx @puradox I just ran into this issue today.

closed 2022-06-04 by petejohanson