zmkfirmware/zmk #1149: Multiple sticky layers not releasing correctly

dlip, 2022-02-27:
- press sticky layer key to go to layer 1
- press sticky layer key to go to layer 2
- press number 1 key in 'a' position
- press a key

expected output:

1a

actual output:

11

okke-formsma, 2022-02-27:
Interesting. It's probably caused by this code:

[image]

This code is in a loop, and it causes the second sticky key not to release. The return should be removed and all items in the loop should be processed.

dlip, 2022-02-28:
@okke-formsma I just tested your fix and it works beautifully! :clap: Thank you so much for helping so quickly! :pray:

dlip, 2022-02-28:
I may have found another small bug, see if you can reproduce:

- press sticky layer key to go to layer 1
- press sticky control key
- press bbbbb quickly

this results in multiple ctrl+b when it should just be 1

dlip, 2022-02-28:
hmm, maybe you can ignore my last bug, I've re-flashed and its not happening for me anymore

dlip, 2022-03-01:
I have found another related issue:

- press sticky layer key to go to layer 1
- hold sticky layer key to go to layer 2 (keep holding)
- press number 1 key in 'a' position
- release sticky layer 2 key
- press 'a' key

expected output:

1a

actual output:

11 (you can keep pressing a to output 1 until the sticky timeout completes)

closed 2022-03-16 by petejohanson