zmkfirmware/zmk #1161: Right side of split keyboard not working

strelov1, 2022-03-05:
If I load on my right side left firmware- it wokred, but refused work with right side

I have nice-nano-v2 and Jorne

https://github.com/strelov1/zmk-config

tabbeber, 2022-03-05:
Have you reset both the halves at the same time to pair them together? The firmware should make the right half not send data through USB as far as I know, so if that's what you're testing it's working as intended.
If you struggle pairing them I reccomend you download the [nice_nano_v2-settings_reset-zmk ](https://github.com/zmkfirmware/zmk/actions/runs/1935069945) and flash that first on both halves, then reflash the left and right firmware on each half.

strelov1, 2022-03-05:
Thanks a lot, reset setting firmware helped me

closed 2022-03-05 by strelov1