zmkfirmware/zmk #1178: [Suggestion] Reuse Github CI definition for zmk_config repositories

elagil, 2022-03-21:
Currently, each zmk_config repository that a user creates, includes its own Github actions pipeline definition.

As described in [this document](https://docs.github.com/en/actions/using-workflows/reusing-workflows), the job definition could be imported from a global place, instead of copying it each time. Thus, it would remain up to date and would not have to be adjusted manually (e.g. for things like [these](https://zmk.dev/blog/2022/03/08/zephyr-3-0-upgrade-prep/)). It looks like these job definitions are always the same, and if not, the imported job can be extended.

If this suggestion makes sense to you, I can create a pull request for this functionality.

joelspadin, 2022-03-22:
Seems like a good idea to me now that we pull configuration from build.yaml in the root of the repo instead of from the workflow filed directly.

elagil, 2022-03-22:
I made two PRs. 

1. Add a `build-user-config.yaml` that closely resembles the current template for user configurations, but with an added header
2. Update the template, so that it uses the new reusable workflow

closed 2022-04-19 by petejohanson