zmkfirmware/zmk #1185: Key repeat for &mt or &lt tap not working?

kekscode, 2022-03-24:
I come from QMK where i use this behaviour:

I tap+hold the LGUI for LGUI or I tap it for getting SPACE using `&mt`. This works in QMK.
In QMK, i can also do the sequence: Tap, tap+hold and i get repeated SPACE . I can't get this working in ZMK. Neither for `&mt` nor for `&lt` (see config details below).

In ZMK i always end up just getting a second SPACE without a repeat and LGUI gets activated. This is of course not wrong, but other keys like ordinary `&kp A` work when i hold them (they get repeated) and i expected that this behaviour i am used to starts working when i set the `&key_repeat` feature.

My config:

The above mentioned SPACE keys:

- `&mt LGUI SPACE`
- `&lt SPECIAL SPACE` (SPECIAL == 1)

To make it work i played around with:

```
&key_repeat {
        usage-pages = <HID_USAGE_KEY HID_USAGE_CONSUMER>;
};
```

Also retro tap is activated:

```
&lt {
        retro-tap;
};

&mt {
    retro-tap;
};
```

The docs are rather short: https://zmk.dev/docs/behaviors/key-repeat and miss the use case or i don't understand them correctly. Of course i would love to contribute to improve the docs if there is a good workaround, because i think others may have the same issue?

caksoylar, 2022-03-24:
Hi, have you seen the `quick-tap-ms` property in the docs? https://zmk.dev/docs/behaviors/hold-tap/#quick-tap-ms It addresses exactly this need. Like noted here, this behavior is opt-in in ZMK as opposed to opt-out in QMK.

kekscode, 2022-03-24:
Thank you @caksoylar for the fast help. I didn't understand that `quick-tap-ms` would help me with my use case, but thanks to you i was able to fix my issue.

I have another related issue or edge-case (see "FIXME" in my `corne.keymap` below), but i can live with it for now. Very happy so far with my wireless crkbd and that it almost works as before with QMK/VIA/VIAL settings now.

For future reference, this is my current configuration:

```
#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>
#include <dt-bindings/zmk/outputs.h>

// Layers
#define DEFAULT 0  // Typing words and stuff
#define SPECIAL 1  // Typing weird characters
#define NAVI    2  // Arrows, Paging and arabic numbers
#define FUNC    3  // Microsoft loves those
#define BLUE    4  // ZMK BLE controls ("Look ma, no cables!")

// Modifiers + Hotkeys
#define TMUX LC(BSLH)

// Properties
&lt {
        //retro-tap; // FIXME: LGUI+TAB for tab switching works. But LGUI+mouseclick does not work correctly when retro-tap is active.
        quick-tap-ms = <199>;  // https://github.com/zmkfirmware/zmk/issues/1185
};
&mt {
        //retro-tap; // FIXME: LGUI+TAB for tab switching works. But LGUI+mouseclick does not work correctly when retro-tap is active.
        quick-tap-ms = <199>;  // https://github.com/zmkfirmware/zmk/issues/1185
};

// Keymap
/ {
        keymap {
                compatible = "zmk,keymap";
                default_layer {
                        bindings = <
   &lt BLUE ESC  &kp Q  &kp W  &kp E  &kp R  &kp T    &kp Y  &kp U  &kp I      &kp O    &kp P     &kp BSPC
   &lt NAVI TAB  &kp A  &kp S  &kp D  &kp F  &kp G    &kp H  &kp J  &kp K      &kp L    &kp SEMI  &kp ENTER
   &kp LSHFT     &kp Z  &kp X  &kp C  &kp V  &kp B    &kp N  &kp M  &kp COMMA  &kp DOT  &kp FSLH  &kp RSHFT

               &kp LCTRL  &mt LGUI SPACE  &kp LALT    &kp TMUX  &lt SPECIAL SPACE  &lt FUNC F1
                        >;
                };
                special_layer {
                        bindings = <
   &kp ESC    &kp EXCL  &kp AT     &kp HASH  &kp DLLR  &kp PRCNT    &kp CARET  &kp AMPS  &kp ASTRK  &kp MINUS  &kp PLUS  &kp DEL
   &kp TAB    &none     &kp UNDER  &kp LBRC  &kp LPAR  &kp SQT      &kp DQT    &kp RPAR  &kp RBRC   &kp MINUS  &kp PIPE  &kp ENTER
   &kp LSHFT  &none     &none      &kp LT    &kp LBKT  &kp TILDE    &kp GRAVE  &kp RBKT  &kp GT     &none      &kp BSLH  &kp RSHFT

                                   &kp LCTRL  &kp LGUI  &kp LALT    &none  &none  &none
                        >;
                };
                navi_layer {
                        bindings = <
    &none      &kp PG_UP  &kp UP    &kp PG_DN  &none  &none    &kp PLUS   &kp N7  &kp N8  &kp N9  &kp DEL    &kp BSPC
    &none      &kp LEFT   &kp DOWN  &kp RIGHT  &none  &none    &kp EQUAL  &kp N4  &kp N5  &kp N6  &kp ASTRK  &kp ENTER
    &kp LSHFT  &none      &none     &none      &none  &none    &kp MINUS  &kp N1  &kp N2  &kp N3  &kp FSLH   &kp RSHFT

                                        &none  &none  &none    &kp SPACE  &kp N0  &kp DOT
                        >;
                };
                func_layer {
                        bindings = <
   &kp F1  &kp F2  &kp F3  &kp F4  &kp F5  &kp F6    &kp F7  &kp F8  &kp F9  &kp F10  &kp F11  &kp F12
   &none   &none   &none   &none   &none   &none     &none   &none   &none   &none    &none    &none
   &none   &none   &none   &none   &none   &none     &none   &none   &none   &none    &none    &none

                              &none  &none  &none    &none  &none  &none
                        >;
                };
                bt_layer {
                        bindings = <
   &none  &bt BT_SEL 0  &bt BT_SEL 1  &bt BT_SEL 2  &bt BT_SEL 3  &bt BT_SEL 4    &none  &none  &none  &none  &none  &bt BT_CLR
   &none  &none         &none         &none         &none         &none           &none  &none  &none  &none  &none  &none
   &none  &none         &none         &none         &none         &out OUT_TOG    &none  &none  &none  &none  &none  &none

                                                           &none  &none  &none    &none  &none  &none 
                        >;
                };
        };
};
```

closed 2022-03-24 by kekscode