zmkfirmware/zmk #1189: nice!nano won't connect (authentication failed)

jwav, 2022-03-27:
I'm using Ubuntu 21.10 and a nice!nano v2. West (installed via pip) version: v0.12.0

After "flashing" the nice!nano with the drag&drop method with an a priori valid UF2 (see below), I can see in the Bluetooth devices list. But when trying to connect with `btdevice`, I get this error : 
```
$ btdevice -c QAZ
Connecting to: QAZ
Error: GDBus.Error:org.bluez.Error.AuthenticationFailed: Authentication Failed
```

Any idea what's causing this?

Additional notes:
This problem arises regardless of how the UF2 is produced (GitHub Actions or locally build) and no error has ever arisen when compiling with this command : `$ west build -p -b nice_nano -- -DSHIELD=qaz`

caksoylar, 2022-04-03:
Did you connect to another device on the same BT profile (on the keyboard)? See [the Bluetooth docs page](https://zmk.dev/docs/behaviors/bluetooth/) for details on how BT profiles and pairing works in ZMK if you aren't familiar.

You can also get help faster and interactively on [the ZMK Discord server](https://zmk.dev/community/discord/invite) if you prefer.

eXsoR65, 2022-04-11:
I had the same issue on a Mac, as @caksoylar was hitting this may be that the devices tired connect on the same profile. I fixed it by resetting all profiles, turning off bluetooth all device except the one I want to pair, pair to the device, and repeat it for all other devices.

jwav, 2022-04-11:
I've just tried the following protocol: 
- 1 laptop and 1 nice!nano being the only bluetooth devices in the vincinity
- power up the nice!nano via a data-less USB line
- on the laptop, use a graphical tool to discover the nice!nano and attempt pairing
- result: pairing failed with the same error pattern described in the first post.

It seems to be an issue unaddressed by @caksoylar 's and @eXsoR65 's posts, if I didn't misunderstand them.

So far I've neglected one possibility because it seemed non pertinent: Does the fact that the nice!nano is powered by its USB connector affect connectivity? Although I don't see why this would matter, you never know until you check the schematics. I'll do this later today.

Ah, and also: it's not connected to a keyboard. But I don't see why this would prevent pairing...

caksoylar, 2022-04-11:
Just to double check, are you aware of how ZMK works with multiple BT profiles on the keyboard? In your message you haven't mentioned clearing or switching profiles.

> Does the fact that the nice!nano is powered by its USB connector affect connectivity?

There is USB detection logic in ZMK which sometimes fails to detect when a device is power-only or not. But that wouldn't prevent BT pairing and connecting, instead it should connect successfully but try to send keystrokes over USB instead of BT. See the note at https://zmk.dev/docs/behaviors/outputs/#summary.

jwav, 2022-04-11:
Got it for the USB thing.
Concerning the BT profiles: I've carefully read the article you linked, but I'm not sure I understand what I could do to see if it's a profile problem.
Is there a way to browse through BT profiles without simulating a `BT_CLR` or `BT_NEXT` press?
If not, I guess I'll have to bodge something to simulate these key presses, but even then I'm not sure what I should do with them.

Any pointers on what to do regarding profiles?

eXsoR65, 2022-04-11:
> Is there a way to browse through BT profiles without simulating a `BT_CLR` or `BT_NEXT` press? If not, I guess I'll have to bodge something to simulate these key presses, but even then I'm not sure what I should do with them.
> 
> Any pointers on what to do regarding profiles?

The "profiles" is bluetooth profile that saves your computer's hardware id etc.. on the Microcontroller's memory. The only way to change/switch profiles or clear aka reset a profile is via key mapping. 

In your keymap add the following. 

1. Add the header for bluetooth here other headers are defined. 

`#include <dt-bindings/zmk/bt.h>`

2. then somewhere on your keymap, map the key codes for related to Bluetooth Profiles. 

To clear a profile
`&bt BT_CLR` 

To switch to the next profile.
`&bt BT_NEXT` , `&bt BT_PRV` ,  or  Selecting a profile `%bt BT_SEL 0` (note you can have up to 5 profile so BT_SEL 0 - 4)

For an example have a look at this keymaps as an example: 
https://github.com/zmkfirmware/zmk/blob/main/app/boards/shields/corne/corne.keymap

jwav, 2022-04-11:
Thanks mate, I can't try this right now but it's on the top of the list.

S3NTYN3L, 2022-04-17:
Thank you, @caksoylar for the hint and thank you, @eXsoR65, for the hand-holding!
I was wracking my brain for about two hours messing with BT front/back-end configs, checking solder joints, etc thinking that was the reason for my constant disconnect/reconnect issue.
I ended up shorting the pads (switches not arrived yet) for the corresponding layer change and BT_CLR keys five times, reenabled BT, reset the N!N and it instantly paired.

BenBlumer, 2023-11-17:
> I had the same issue on a Mac, as @caksoylar was hitting this may be that the devices tired connect on the same profile. I fixed it by resetting all profiles, turning off bluetooth all device except the one I want to pair, pair to the device, and repeat it for all other devices.

This worked for me. Clearing all the other profiles was the ticket unfortunately.  Is this a bug? Is there any explanation for this?

BenBlumer, 2023-11-17:
Oh! 

So -- I dual boot Ubuntu and Windows 11. I had my keyboard paired with W11 (as well as a phone and a tablet). The issue arose when I was trying to pair it with Ubuntu. Clearing all the profiles helped. I then booted W11 and tried to re-pair. It wouldn't do it. This time, I only cleared the profile for the Ubuntu connection. All of a sudden, it would pair.

It seems like ZMK doesn't want to pair to two OSes on the same device (or with the same bluetooth MAC address?).

Is this a fundamental bluetooth issue or a ZMK issue?