zmkfirmware/zmk #1190: Disable Auto shift when LGUI is pressed (for copy or other actions)

Vasilis, 2022-03-27:
Hi, everyone! I recently enabled auto-shift on my Corne-ish Zen. That works fine except for the copy - paste commands where instead of `LGUI+C`, what is sent is `LGUI+SHIFT+C` if I accidentally hold it for longer than `tapping-term-ms`.

Is it possible to **prevent the hold-tap behavior altogether when LGUI (or any other key) is pressed before**? I tried the following unsuccessfully:
```
        behaviors {
                as: auto_shift {
                        compatible = "zmk,behavior-hold-tap";
                        label = "AUTO_SHIFT";
                        #binding-cells = <2>;
                        tapping-term-ms = <210>;
                        quick-tap-ms = <0>;
                        flavor = "tap-preferred";
                        bindings = <&kp>, <&kp>;
                };
                modcp: mod_copy {
                        compatible = "zmk,behavior-mod-morph";
                        label = "MOD_COPY";
                        #binding-cells = <0>;
                        bindings = <___C___>, <MAC_COPY>;
                        mods = <(MOD_LGUI | MOD_RGUI)>;
                };
                modpst: mod_paste {
                        compatible = "zmk,behavior-mod-morph";
                        label = "MOD_PASTE";
                        #binding-cells = <0>;
                        bindings = <___V___>, <MAC_PASTE>;
                        mods = <(MOD_LGUI | MOD_RGUI)>;
                };
        };
```

MatCyg, 2022-03-28:
Here you go for the copy (remember to use mod morph in the keymap):

```c
as: auto_shift {
        compatible = "zmk,behavior-hold-tap";
        label = "AUTO_SHIFT";
        #binding-cells = <2>;
        tapping-term-ms = <210>;
        quick-tap-ms = <0>;
        flavor = "tap-preferred";
        bindings = <&kp>, <&kp>;
};
modcp: mod_copy {
        compatible = "zmk,behavior-mod-morph";
        label = "MOD_COPY";
        #binding-cells = <0>;
        bindings = <&as LS(C) C>, <&kp C>;
        mods = <(MOD_LGUI | MOD_RGUI)>;
};

use in keymap:
&modcp          
```

Vasilis, 2022-03-28:
Thanks for your time, @MatCyg! Unfortunately, this still does not work on (on a Corne-ish Zen). 

I assigned `&modcp` to `C` in the keymap and, still, if I press `LGUI`+`C` for longer than  `tapping-term-ms`, it will still output the autoshift (`&as LS(C) C`) with shift in the mix... :/

Depending on the active application, the behavior is the not desired one (for instance on a browser it brings up the developer section)

MatCyg, 2022-03-28:
Sorry, I misunderstood. I don't think you can do it with mod morph, but it is fairly easy with a macro. An idea is to switch to normal alpha layer without auto shift when you press CMD. This way you do it for all your keys, not only copy/paste.

```c
ZMK_MACRO(cmdAlpha,
    wait-ms = <0>;
    tap-ms = <0>;
    bindings
        = <&macro_press &mo ALPHA &kp LCTL>,
          <&macro_pause_for_release>,
          <&macro_release &mo ALPHA &kp LCTL> ;
)
```

Vasilis, 2022-04-06:
Thank you, @MatCyg ! Will try it out!

elismaga, 2023-02-22:
Did this work? I don't quite understand how that solution works or how to implement it. 
In QMK the default behavior is what I am looking for which is what the OP described. In order to get the default behavior of ZMK you have to enable AUTO_SHIFT_MODIFIERS. I was hoping ZMK had a setting like this to reverse the default behavior.
[AUTO_SHIFT_MODIFIERS](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_auto_shift.md#modifiers)

elismaga, 2023-02-22:
@MatCyg What if you set "hold-trigger-key-positions" to an empty array in your auto shift definition?

feerrenrut, 2023-03-11:
@elismaga I think the suggestion looks like this: https://github.com/feerrenrut/ergodox-zmk-config/commit/e75d0cea86e6cf5f2d5a66d030fb53de26f00596

One issue is that a macro needs to be created for each mod key that used. I'll need separate macros for each of the following:
1. left control
1. right control
1. left shift
1. right shift
1. left meta
1. left alt
1. right alt

It would be nice if a macro accepted arguments, then you could make a macro to use like `NO_AUTO_SHIFT(TO_LAYER, LCTRL)`

BhuvanB, 2024-08-27:
@feerrenrut, I'm new to ZMK, and maybe I'm missing some nuance here (or maybe this is new functionality introduced since you commented), but it seems https://zmk.dev/docs/keymaps/behaviors/macros#parameterized-macros provides you with the ability to use macros with up to two arguments.