zmkfirmware/zmk #1194: `CONFIG_BT_MAX_CONN` unused in Bluetooth behavior

kurtis-lew, 2022-03-29:
Shown below, the docs show that adjusting the number of Bluetooth profiles for a keyboard is done through editing the value of `CONFIG_BT_MAX_CONN` in a given keyboard's `.conf` file.
[image]

This issue was discovered in the attempt to limit the number of Bluetooth profiles to 2. Given the following code in my `iris.conf`:
```
CONFIG_ZMK_SLEEP=y
CONFIG_ZMK_USB_LOGGING=y
CONFIG_BT_MAX_CONN=2
```

The Iris is able to store 5 different Bluetooth profiles. The only way to ensure that the number of profiles is limited to exactly 2 profiles is through the following `.conf`:
```
CONFIG_ZMK_SLEEP=y
CONFIG_ZMK_USB_LOGGING=y
CONFIG_BT_MAX_PAIRED=3
```

The following images show that `CONFIG_BT_MAX_CONN` is unused in both `ble.h` and `ble.c`, while `CONFIG_BT_MAX_PAIRED` is used to determine the value of `ZMK_BLE_PROFILE_COUNT`.

[image]
[image]

`ble.h`
[image]

Since `ZMK_BLE_PROFILE_COUNT` is the config value primarily used in `ble.c`, it would also seem that the following lines are unnecessary in `ble.c`:
[image]

In short:
- Either update docs to show that `CONFIG_BT_MAX_PAIRED` is the variable that should be changed _or_ update `ble.c` to implement `CONFIG_BT_MAX_CONN`
- If the former solution is chosen, ensure users are aware that on split keyboards `ZMK_BLE_PROFILE_COUNT` is equal to `CONFIG_BT_MAX_PAIRED - 1` on the central half

joelspadin, 2022-03-29:
Don't both values need to be updated together? Because ZMK keeps a connection open on all paired profiles for fast profile switching, if you can pair N (+1 for split) profiles, you need to support the same number of active connections.

kurtis-lew, 2022-03-29:
If I'm understanding correctly, do you mean to say that both `CONFIG_BT_MAX_PAIRED` and `CONFIG_BT_MAX_CONN` need to be updated togther? I'm unsure about that. I've updated the original post with labels on images, but Fig. 1 suggests that `CONFIG_BT_MAX_CONN` isn't necessary driver-wise.

joelspadin, 2022-03-29:
Bluetooth is mostly controlled by Zephyr. ZMK tells it what to do, but the actual BT stack is in `zephyr/subsys/bluetooth`, where both `CONFIG_BT_MAX_PAIRED` and `CONFIG_BT_MAX_CONN` are used

kurtis-lew, 2022-03-29:
I see what you're saying. In that case, would it be fair to say that this is just a documentation issue?

joelspadin, 2022-03-29:
I think so. Note that ZMK's Kconfig file sets both values to be the same in all cases. If our docs don't call out both settings, they probably should.

kurtis-lew, 2022-03-29:
Sounds simple enough, thanks for the review. I'll get around to putting out a PR in the next day or so.

closed 2022-03-29 by kurtis-lew