zmkfirmware/zmk #1207: Key repeat cannot be rolled

andrewjrae, 2022-04-03:
Currently key repeat cannot be rolled with another key. For example: `G-down -> key_repeat-down -> G-up -> key_repeat-up` results in `g`, when it should result in `gg`. This is important functionality for those trying to use key repeat in everyday typing to reduce same finger bigrams.

This is also somewhat related to #1076, since the key code needs to be released before being tapped again.

It may be possible to fix this at the core by enforcing that a release is sent before pressing a code that is already pressed (this should also solve the problem in #1076). However, I'm not sure this is always desirable behavior (and could  potentially end up in erroneous releases being sent), and so for now I'll implement a patch just for key repeat. If others think that this standalone patch is preferred, I'll put up a PR.

gaoDean, 2022-08-04:
Used it for a few minutes, 10/10. Thank you so much!

ze-or, 2022-12-22:
since I replaced  ZMK_EV_EVENT_CAPTURED with ZMK_EV_EVENT_HANDLED crashing has stopped.

The rolling issue for triple presses of the same key is the only remaining issue I can tell.

andrewjrae, 2023-05-30:
FYI @ze-or I've implemented a new patch that should fix your triple roll case, by patching at the core. I have yet to test this for very long, but so far it seems to work well.

@petejohanson, I think this fix is more along the lines of what we'd want, and I _think_ I put it in a good spot.

If anyone else has success with this let me know and I can put up a PR!

caksoylar, 2023-06-05:
It works fine for me, e.g. to roll `+=` or `-_-` which weren't possible before, thanks @andrewjrae!

commit 933e369d (2023-10-02): feat(core): Adding pre-release for keys that were already pressed.

This fixes #1207 and #1076 (and maybe more?).
closed 2023-10-02 by petejohanson