zmkfirmware/zmk #1235: Entering sleep doesn't turn off RGB underglow since Zephyr 3.0 update

kjerstadius, 2022-04-10:
Since the Zephyr 3.0 update sleep seems to behave slightly differently, such as no longer activating sleep when a USB cable is connected (which I find great). However, one regression I've found is that RGB underglow no longer turns off when entering sleep mode, as it used to before the update.

**Steps to reproduce**
* Enable RGB underglow
* Enable sleep mode
* Let keyboard enter sleep

Expected behavior: LEDs are turned off
Actual behavior: LEDs are lit even though the keyboard has entered sleep mode

I have verified via debug logging that the keyboard is indeed entering sleep, so that isn't the issue. The actions taken before entering sleep must have changed somehow.

The board I'm experiencing this on is a Kyria v1.4, using Nice!Nano v2s and the following config:

```
# Linear encoders
CONFIG_EC11=y
CONFIG_EC11_TRIGGER_GLOBAL_THREAD=y

CONFIG_ZMK_DISPLAY=y
CONFIG_ZMK_WIDGET_WPM_STATUS=n

CONFIG_ZMK_RGB_UNDERGLOW=y
CONFIG_WS2812_STRIP=y
CONFIG_ZMK_RGB_UNDERGLOW_EXT_POWER=n
CONFIG_ZMK_RGB_UNDERGLOW_ON_START=y

CONFIG_ZMK_SLEEP=y
CONFIG_ZMK_IDLE_TIMEOUT=60000
CONFIG_ZMK_IDLE_SLEEP_TIMEOUT=1800000

CONFIG_ZMK_KSCAN_DEBOUNCE_PRESS_MS=6
CONFIG_ZMK_KSCAN_DEBOUNCE_RELEASE_MS=2

CONFIG_ZMK_USB_LOGGING=n
```

I tried changing `CONFIG_ZMK_RGB_UNDERGLOW_EXT_POWER` to `y` to see if that had anything to do with it, but it makes no difference AFAICT.

edmondhuang, 2022-04-25:
V3.0 - it is real issue, tested, RGB doesn't turn off automatically while they come to deep sleep. Previous version is working fine.

petejohanson, 2022-05-26:
Can you please test again? @Nicell pushed a fix for this in #1291 that should have resolved it.

kjerstadius, 2022-05-29:
I can confirn that the issue is absent from the current main. I'll close this issue. Thanks!

closed 2022-05-29 by kjerstadius