zmkfirmware/zmk #1237: USB_DEVICE_STACK Dependency loop

jwav, 2022-04-11:
I just did a pull for zmk and a `pip install --upgrade west`.

I'm trying to compile zmk-firwmare with west with this command: `$ west build -p -b nice_nano -- -DSHIELD=qaz`

and getting the output below.
Apparently it's the following dependency loop:
- USB_DEVICE_STACK depends on USB_DEVICE_DRIVER
- USB_DEVICE_DRIVER depends on USB_NRFX
- USB_NRFX depends on USB_DEVICE_STACK

Any idea on how to fix it?

```
-- west build: making build dir /home/vee/github/zmk/app/build pristine
-- west build: generating a build system
-- Using keymap file: /home/vee/github/zmk/app/boards/shields/qaz/qaz.keymap
Including boilerplate (Zephyr base): /home/vee/github/zmk/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/vee/github/zmk/app
-- Zephyr version: 2.5.0 (/home/vee/github/zmk/zephyr)
-- Found Python3: /usr/bin/python3.9 (found suitable exact version "3.9.7") found components: Interpreter 
-- Found west (found suitable version "0.13.0", minimum required is "0.7.1")
-- git describe failed: fatal: No tags can describe 'f2108f1646b964adffb9c7ed74641c828fef2a44'.
Try --always, or create some tags.;
   BUILD_VERSION is left undefined
-- Board: nice_nano, Shield(s): qaz
-- Cache files will be written to: /home/vee/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.12.4 (/home/vee/.local/zephyr-sdk-0.12.4)
-- Found dtc: /home/vee/.local/zephyr-sdk-0.12.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/vee/github/zmk/app/boards/arm/nice_nano/nice_nano.dts
-- Found devicetree overlay: /home/vee/github/zmk/app/boards/shields/qaz/qaz.overlay
-- Found devicetree overlay: /home/vee/github/zmk/app/boards/shields/qaz/qaz.keymap
-- Generated zephyr.dts: /home/vee/github/zmk/app/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/vee/github/zmk/app/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/vee/github/zmk/app/build/zephyr/include/generated/device_extern.h
Parsing /home/vee/github/zmk/app/Kconfig
**/home/vee/github/zmk/zephyr/scripts/kconfig/kconfig.py: Dependency loop**
===============

USB_DEVICE_STACK (defined at /home/vee/github/zmk/app/Kconfig:519, subsys/usb/Kconfig:6), with definition...

config USB_DEVICE_STACK
	bool
	default y if HAS_HW_NRF_USBD

menuconfig USB_DEVICE_STACK
	bool "USB Device Support"
	select HWINFO
	depends on USB_DEVICE_DRIVER || ARCH_POSIX
	help
	  Enable USB device stack.

(select-related dependencies: ZMK_USB || ZMK_USB_LOGGING)

...depends on USB_DEVICE_DRIVER (defined at drivers/usb/device/Kconfig:8), with definition...

config USB_DEVICE_DRIVER
	bool
	depends on USB

(select-related dependencies: (USB_DW && USB) || (USB_DC_STM32 && SOC_FAMILY_STM32 && USB) || (USB_DC_SAM0 && SOC_FAMILY_SAM0 && USB) || (USB_DC_SAM && (SOC_SERIES_SAME70 || SOC_SERIES_SAMV71) && USB) || (USB_NRFX && HAS_HW_NRF_USBD && USB) || (USB_KINETIS && USB) || (USB_DC_NXP_EHCI && HAS_MCUX_USB_EHCI && USB) || (USB_NATIVE_POSIX && USB))

...depends on USB_NRFX (defined at /home/vee/github/zmk/app/boards/arm/nice_nano/Kconfig.defconfig:11, drivers/usb/device/Kconfig:50), with definition...

config USB_NRFX
	bool
	default y
	depends on USB_DEVICE_STACK && (BOARD_NICE_NANO || BOARD_NICE_NANO_V2)

menuconfig USB_NRFX
	bool "Nordic Semiconductor USB Device Controller Driver"
	default y
	select USB_DEVICE_DRIVER
	select NRFX_USBD
	select NRFX_POWER
	depends on HAS_HW_NRF_USBD && USB
	help
	  nRF USB Device Controller Driver

...depends again on USB_DEVICE_STACK (defined at /home/vee/github/zmk/app/Kconfig:519, subsys/usb/Kconfig:6)
CMake Error at /home/vee/github/zmk/zephyr/cmake/kconfig.cmake:265 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/vee/github/zmk/zephyr/cmake/app/boilerplate.cmake:536 (include)
  /home/vee/github/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/vee/github/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:17 (find_package)

**-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/vee/github/zmk/app/build -S/home/vee/github/zmk/app -GNinja -DBOARD=nice_nano -DSHIELD=qaz**

```

petejohanson, 2022-04-11:
Can you completely remove the build directory and try again? Seems like Zephyr 3.0 upgrade funkiness.

jwav, 2022-04-11:
Just did that, rebuilt, got the same error. Good guess, though.

petejohanson, 2022-04-11:
And you've run `west updated` properly?

caksoylar, 2022-04-11:
Logs say `Zephyr version: 2.5.0 (/home/vee/github/zmk/zephyr)`, so I would make sure you have `git pull`'ed the ZMK repo and `west update`d like Pete mentioned.

jwav, 2022-04-11:
I `git pull`ed, `west update`d, updated cmake as well to the latest version, and now when running `$ west build -b nice_nano -- -DSHIELD=qaz` I get the following output.

Enough for today -_-

```
-- west build: generating a build system
-- Using keymap file: /home/vee/github/zmk/app/boards/shields/qaz/qaz.keymap
Including boilerplate (Zephyr base): /home/vee/github/zmk/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/vee/github/zmk/app
-- Zephyr version: 3.0.0 (/home/vee/github/zmk/zephyr)
-- Found Python3: /usr/bin/python3.9 (found suitable exact version "3.9.7") found components: Interpreter 
-- Found west (found suitable version "0.13.0", minimum required is "0.7.1")
-- Board: nice_nano, Shield(s): qaz
-- Cache files will be written to: /home/vee/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
CMake Error at /home/vee/github/zmk/zephyr/cmake/verify-toolchain.cmake:75 (find_package):
  Could not find a configuration file for package "Zephyr-sdk" that is
  compatible with requested version "0.13.1".

  The following configuration files were considered but not accepted:

    /home/vee/.local/zephyr-sdk-0.12.4/cmake/Zephyr-sdkConfig.cmake, version: 0.12.4

Call Stack (most recent call first):
  /home/vee/github/zmk/zephyr/cmake/app/boilerplate.cmake:519 (include)
  /home/vee/github/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/vee/github/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /snap/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/vee/github/zmk/app/build -S/home/vee/github/zmk/app -GNinja -DBOARD=nice_nano -DSHIELD=qaz
```

caksoylar, 2022-04-11:
Sounds like you might need to install a newer toolchain version as well, following https://zmk.dev/docs/development/setup/#toolchain-installation:

> Could not find a configuration file for package "Zephyr-sdk" that is
>  compatible with requested version "0.13.1".
 > The following configuration files were considered but not accepted:
  >  /home/vee/.local/zephyr-sdk-0.12.4/cmake/Zephyr-sdkConfig.cmake, version: 0.12.4

jwav, 2022-04-11:
> Sounds like you might need to install a newer toolchain version as well, following https://zmk.dev/docs/development/setup/#toolchain-installation:

That fixed it, thanks to this chunk below.
Thanks a lot, @caksoylar 

```
export ZSDK_VERSION=0.13.2
wget -q "https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v${ZSDK_VERSION}/zephyr-toolchain-arm-${ZSDK_VERSION}-linux-x86_64-setup.run" && \
    sh "zephyr-toolchain-arm-${ZSDK_VERSION}-linux-x86_64-setup.run" --quiet -- -d ~/.local/zephyr-sdk-${ZSDK_VERSION} && \
    rm "zephyr-toolchain-arm-${ZSDK_VERSION}-linux-x86_64-setup.run"
```

closed 2022-04-11 by jwav