zmkfirmware/zmk #1260: Layer change hook

saesols, 2022-04-24:
Is it possible to add layer change action hook. For example send key combination to change language. It would be very helpful for a multi-lang users. One can sand lang change before layer activation.

joelspadin, 2022-04-24:
I assume you mean to press some key combination that changes the OS keyboard locale, then to switch into a layer? Could this be done with a macro that sends a key combination and then `&to` to switch layers?

saesols, 2022-04-25:
I don't think so. 

Let's assume there are 3 lang layers with the respective shortcuts:
1. Eng - CTRL+SHIFT+1
2. Ger - CTRL+SHIFT+2
3. Spa - CTRL+SHIFT+3

And a Sym symbol layer with []{} etc. which always use Eng OS layout.

Using macro I don't find the way to return to the language used before one switched to the Sym layer.
 
With the proposed hooks it is possible to send the respective OS lang laout change before every lang layer.

saesols, 2022-05-03:
Link to the same functionality in QMK:
https://github.com/qmk/qmk_firmware/blob/master/docs/custom_quantum_functions.md#layer-change-code-idlayer-change-code

labuut, 2022-11-12:
@saesols, hello! I have the same problem. Have you found any solution?

saesols, 2022-11-12:
@labuut Hi, nope. I've switched to QMK =(

labuut, 2022-11-12:
I think I will follow your example (

jcmkk3, 2022-11-13:
I'm pretty certain that this should be possible with macros and layers. We'd need to see your exact layout to know how to give specific advice, but you should be able to make each of the lang layers also have their own macro that switches to the symbol layer and switches back to the language that you came from.