zmkfirmware/zmk #1273: Keyboard prevents macOS from sleeping

dxmh, 2022-04-27:
macOS users often report their machines being woken from sleep  by their keyboard (causing laptop battery drain, etc).

@petejohanson mentioned [on Discord][1]:

> It's likely the battery status notifications waking it.
> We probably need some work on our end to honor when a device is going to sleep to stop doing that.
> Alternately, enabling ZMK deep sleep should help.
> Since the keyboard would sleep also eventually, and stop waking your MB.

### Workaround

For users wishing to workaround this, deep sleep can be enabled and configured in your `<keyboard>.conf` file ([example][3]):

```sh
CONFIG_ZMK_SLEEP=y # Enable deep sleep support
CONFIG_ZMK_IDLE_SLEEP_TIMEOUT=900000 # Milliseconds of inactivity before entering deep sleep
```

(Note that deep sleep is not yet documented, though it is covered on the [Power configuration][2] page from the work-in-progress PR #722.)

[1]: https://discord.com/channels/719497620560543766/719909884769992755/811654995761692703
[2]: https://deploy-preview-722--zmk.netlify.app/docs/config/power
[3]: https://github.com/dxmh/zmk-config/blob/3097036876e6682b6915eddcf75643391e77fff4/config/a_dux.conf#L2-L3

kanru, 2022-05-16:
Is it possible to change the frequency of the battery status notification being sent? For example I don't need to know battery status per second. Probably hourly is good enough for the purpose of knowing that I need to charge it.

caksoylar, 2022-05-16:
@kanru This seems to be hard-coded to check every minute; you can try to change the timer constant.

petejohanson, 2022-05-16:
> @kanru This seems to be hard-coded to check every minute; you can try to change the timer constant at: https://github.com/zmkfirmware/zmk/blob/main/app/src/battery.c#L107
> 
> 

I'd also be very open to moving that hardcoded value into a Kconfig setting, PRs welcomed.

filterpaper, 2022-05-19:
`CONFIG_ZMK_IDLE_SLEEP_TIMEOUT` is millisecond unit while `CONFIG_ZMK_BATTERY_REPORT_INTERVAL` is in seconds. Aligning both might be helpful for consistency and comparison to set `CONFIG_ZMK_BATTERY_REPORT_INTERVAL` > `CONFIG_ZMK_IDLE_SLEEP_TIMEOUT`

dxmh, 2022-09-09:
Recently I’ve started using this utility to disable Bluetooth on the Mac when the machine goes to sleep:

https://github.com/odlp/bluesnooze

It’s been a great workaround to stop my keyboard from waking my Mac.

tsndr, 2023-07-11:
I'm also encountering the issue, when will this be implemented? I really don't want to have to go back to qmk.

filterpaper, 2023-07-11:
It is already implemented. #1303 was merged last year.

tsndr, 2023-07-11:
But why does my keyboard then still wake up my mac constantly?

filterpaper, 2023-07-11:
You will need to explicitly set the battery level reporting interval longer. Here is a setting for `<shield>.conf` that should work:

```c
CONFIG_ZMK_SLEEP=y
CONFIG_ZMK_IDLE_SLEEP_TIMEOUT=1800000
CONFIG_ZMK_BATTERY_REPORT_INTERVAL=1800
```

tsndr, 2023-07-11:
Thank you so much, it was driving me insane especially since my Apple Watch kept unlocking the Mac when it happened so it didn't even go back to sleep. Also my keyboard doesn't habe a power switch, so I couldn't even turn it off.

caksoylar, 2023-07-11:
You can also disable battery reporting completely with `CONFIG_ZMK_BATTERY_REPORTING=n` if you like.

tsndr, 2023-07-11:
Also a great suggestion, thank you :)

tsndr, 2023-07-13:
> You can also disable battery reporting completely with `CONFIG_ZMK_BATTERY_REPORTING=n` if you like.

I've tried setting `CONFIG_ZMK_BATTERY_REPORTING=n` now, but then it fails to build. Am I doing something wrong or could this be a bug?

Log:

```
[300/314] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
[301/314] Building C object zephyr/kernel/CMakeFiles/kernel.dir/poll.c.obj
[302/314] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mempool.c.obj
[303/314] Linking C static library zephyr/kernel/libkernel.a
[304/314] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /__w/zmk-config/zmk-config/build/zephyr/zephyr_pre0.map 
: && ccache /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/__w/zmk-config/zmk-config/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a  zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/usb/device/class/hid/libsubsys__usb__device__class__hid.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/usb/device/libdrivers__usb__device.a  zephyr/drivers/adc/libdrivers__adc.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/hwinfo/libdrivers__hwinfo.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/drivers/kscan/libzmk__drivers__kscan.a  modules/drivers/sensor/battery/lib..__zmk__app__drivers__sensor__battery.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/thumb/v7e-m/nofp"  -L/__w/zmk-config/zmk-config/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /__w/zmk-config/zmk-config/build/zephyr && /usr/local/lib/python3.8/dist-packages/cmake/data/bin/cmake -E echo
/opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: warning: orphan section `.z_device_POST_KERNELCONFIG_SENSOR_INIT_PRIORITY_' from `modules/drivers/sensor/battery/lib..__zmk__app__drivers__sensor__battery.a(battery_nrf_vddh.c.obj)' being placed in section `.z_device_POST_KERNELCONFIG_SENSOR_INIT_PRIORITY_'
/opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: Undefined initialization levels used.
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /__w/zmk-config/zmk-config/build
Error: Process completed with exit code 1.
```

caksoylar, 2023-07-13:
@tsndr Can you share a link to the failing build (on a public config repo)?

tsndr, 2023-07-13:
https://github.com/tsndr/zmk-config/actions/runs/5536781382

caksoylar, 2023-07-14:
It looks like there were some issues with the setting, I opened a PR here as one way to fix them if you want to give it a try: #1867

You can change your `config/west.yml` to this to test:
```yaml
manifest:
  remotes:
    - name: zmkfirmware
      url-base: https://github.com/zmkfirmware
    - name: caksoylar
      url-base: https://github.com/caksoylar
  projects:
    - name: zmk
      remote: caksoylar
      revision: fix-battery-reporting
      import: app/west.yml
  self:
    path: config
```

tsndr, 2023-07-14:
Alright, with your branch it builds. When first flashing it the right keyboard half didn't work, but a reflash solved that. Now evering seems to be working. In my Bluetooth settings it now says `100%`, which I think is expected and will stay stay that way now.

I will keep an eye on the PR and switch back over once it's merged.

Thank you :)

johnnyBira, 2023-08-03:
My build is failing with the same error as @tsndr. I do not have the `CONFIG_ZMK_BATTERY_REPORTING` setting in my config.
https://github.com/johnnyBira/zmk-config/actions/runs/5740711278

@caksoylar I noticed your PR got merged. Would this fix also resolve my issue, and of so what steps would I need to take in order to apply the fix?

caksoylar, 2023-08-03:
@johnnyBira It's not the same issue the PR is fixing, but a similar one where some feature (in this case the battery widget in the nice!view status screen) is not accounting for the possibility of reporting being disabled. I'll go over that and other possible uses and create another PR when I get a chance.

johnnyBira, 2023-08-04:
Thanks @caksoylar, appreciate it 🙏

johnnyBira, 2023-09-10:
Today I tried to get around this issue by simply disabling the battery widget, I instead bumped into another issue:

https://github.com/johnnyBira/zmk-config/actions/runs/6135817090/job/16649960026
```sh
warning: Deprecated symbol BT_LL_SW_LLCP_LEGACY is enabled.
```

Is there a fix on the way or is there any workaround? My builds used to pass and it makes me wonder if it would be possible to build for a different ZMK version until these issues have been resolved in the latest version.

joelspadin, 2023-09-10:
> ```shell
> warning: Deprecated symbol BT_LL_SW_LLCP_LEGACY is enabled.
> ```
> 
> Is there a fix on the way or is there any workaround? My builds used to pass and it makes me wonder if it would be possible to build for a different ZMK version until these issues have been resolved in the latest version.

That is just a warning. The error that caused the build to fail looks to be

> /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(status.c.obj):/__w/zmk-config/zmk-config/zmk/app/boards/shields/nice_view/widgets/status.c:281: multiple definition of `widget_layer_status_mutex'; app/libapp.a(layer_status.c.obj):/__w/zmk-config/zmk-config/zmk/app/src/display/widgets/layer_status.c:51: first defined here

> /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(status.c.obj):/__w/zmk-config/zmk-config/zmk/app/boards/shields/nice_view/widgets/status.c:281: multiple definition of `widget_layer_status_work'; app/libapp.a(layer_status.c.obj):/__w/zmk-config/zmk-config/zmk/app/src/display/widgets/layer_status.c:51: first defined here

> /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(status.c.obj):/__w/zmk-config/zmk-config/zmk/app/boards/shields/nice_view/widgets/status.c:284: multiple definition of `zmk_event_sub_widget_layer_statuszmk_layer_state_changed'; app/libapp.a(layer_status.c.obj):/__w/zmk-config/zmk-config/zmk/app/src/display/widgets/layer_status.c:54: first defined here

It seems to be trying to build both the standard ZMK layer status widget as well as the nice!view status screen.

johnnyBira, 2023-09-10:
@joelspadin okey. Would you say it's a bug or something wrong with my config?

johnnyBira, 2023-09-10:
@joelspadin Okey! Would you say it's a bug or something wrong that's wrong with my config?
I'm not really sure where to start looking,

caksoylar, 2023-09-10:
You cannot use [these widget Kconfigs](https://github.com/johnnyBira/zmk-config/blob/main/config/splitkb_aurora_corne.conf#L3) they are for the stock status screen, and the custom status screen of `nice_view` doesn't respect them. You can turn off the custom status screen following [these instructions](https://github.com/zmkfirmware/zmk/blob/main/app/boards/shields/nice_view/README.md#disable-custom-widget), _then_ use `CONFIG_ZMK_WIDGET_BATTERY_STATUS=n` to turn off the battery widget.

johnnyBira, 2023-09-12:
Thanks @caksoylar , my build is passing again 👌

kelchm, 2023-09-23:
Has anyone had any success with solving this issue without _entirely_ disabling battery reporting? 

While applying the following configuration has drastically improved the situation, I'm still seeing my MBP wake from sleep intermittently.

> ```c
> CONFIG_ZMK_SLEEP=y
> CONFIG_ZMK_IDLE_SLEEP_TIMEOUT=1800000
> CONFIG_ZMK_BATTERY_REPORT_INTERVAL=1800
> ```