zmkfirmware/zmk #1284: Layer documentation is unclear

notfed, 2022-05-03:
The layer documentation [1] does not clearly explain how layers are numbered. Also, it seems to give conflicting information: it explicitly gives a metaphor of layers as numbered by "[reverse-indexed] top-to-bottom" [1], but then the very first example of a keymap uses the exact opposite order with no clarification. [2]

For example: since the "layers" page is where users will go to if they're confused about how to define layers, it would help to say something like:

> Layers will be automatically assigned numbers, beginning with 0 (usually referred to as the 'default layer'), and increasing sequentially by 1 in the order they are defined in the keymap file. <<It would also help to clarify again the confusing fact that layers must be defined in the opposite order as was just explained in [1]>>

But AFAICT no such explanation exists, and so the ZMK user is left to head scratching and assumptions based off the very small number of examples.

And regarding the examples: consider that this is the official ZMK "Layers" documentation, and it gives only a measly 2-layer example. Adding an example with 3 or more will drive home the fact that layers are sequentially ordered; something that I, as a new ZMK user setting up a board for the first time, am still confused about.

[1] https://zmk.dev/docs/features/keymaps
[2] https://zmk.dev/docs/behaviors/layers

butterdori, 2022-05-10:
This is helpful. I couldn't find out how ZMK found the correct layer. Thanks to you, it seems that they are referred to in the order they are written out in the keymap, and therefore the order of the layers in the keymap is important. I agree that an example with more than 3 layers would be more helpful.

zaddok, 2022-12-30:
I am here from google, because the documentation is unclear. More examples would indeed be helpful.

Auxnon, 2023-02-20:
Seconded, this is the first place I've found that clarified that the naming in the keymap file for layers is arbitrary and it's the order named that dictates their layer index. I was under the impression the layer name had to be defined elsewhere and couldn't find it. An example with 4 or more layers and clarifying this concept would be appreciated.

joelspadin, 2024-01-06:
> Wouldn't it help if behaviors like &mo could reference labels by name/label?

It would, but that's not possible with the devicetree format, at least not without requiring something like

```dts
/ { 
  behaviors {
    mo_nav: mo_nav {
      compatible = "zmk,behavior-momentary-layer";
      #binding-cells = <0>;
      layer = <&nav>;
    };
    mo_media: mo_media {
      compatible = "zmk,behavior-momentary-layer";
      #binding-cells = <0>;
      layer = <&media>;
    };
  };
  
  keymap {
    compatible = "zmk,keymap";
    default_layer {
      bindings = <&mo_nav &mo_media>;
    };
    nav: nav_layer {
      bindings = <&trans &kp HOME>;
    };
    media: media_layer {
      bindings = <&kp C_PLAY_PAUSE &trans>;
    };
  };
};
```

where you'd have to manually define a new behavior for each layer you want to go to. The parameters following a behavior in a `bindings` list can only be numbers.

PR #2390 (2024-08-01): mini-feat(docs) Improvements to the layer documentation
Rephased some things and added some clarification. The goal is to resolve #1284.
commit 9e7fcde8 (2024-08-01): feat(docs): Improve the layers-related documentation (#2390)

Fixes #1284
closed 2024-08-01 by caksoylar