zmkfirmware/zmk #1292: Feature request: Soft shutdown of keyboard

dxmh, 2022-05-07:
Lots of users request the ability to power off their keyboard (in the absence of a physical power switch). 

The reason behind most of the requests is so that the keyboard can be transported without sending keystrokes and/or draining battery. However, it's also just nice to be able to turn things off when they're not being used. 🙂

R3XP, 2022-05-07:
How would that work in theory? Would it shut down until a certain keycombo is pressed? If yes, how often does the mcu scan for the combo? If no, is there another way to start it again? Pluging it into something maybe? Or a designated button to restart it? Maybe a reset button?

{_just brainstorming out loud here ^^_}

dxmh, 2022-05-07:
Good question. There was some discussion about this [on Discord](https://discord.com/channels/719497620560543766/719909884769992755/970615483109892156) a few days ago. Here are some relevant snippets to aid the brainstorming:

> **hxy**: 
> How would you power-on the keyboard after a soft shutdown?

> **nef**: 
> reset switch would work...

> **petejohanson**: 
> We could have a software shutdown, that stops BLE and reduces scan frequency.
> So we could still detect key presses, but much more slowly.
> Or we set a flag and go into deep sleep, so if a key press does wake us, still stay in "software lockout mode"

R3XP, 2022-05-07:
Hmm I see. What do you think about a user defined combo where the mcu would only ever scan one of the columns of the combo so that there wouldnt be as much activity as usally but when theres any key pressed in that column it would start scanning for the whole combo again?

manna-harbour, 2022-06-22:
Also see https://github.com/zmkfirmware/zmk/issues/405.

freqmod, 2022-10-02:
Hi, i did a 2 hour stab at this (instead of figuring how to soldier a power switch to my corne helix), and found a solution that seems to work for me. I haven't measured the current draw, but i do not get any presses until reset i pressed.

The following patches are needed (if someone wants to make this into a proper upstream feature it would be great. feel free, and use the MIT license. It would be great if you'd mention my name in the commit message):
(Based on zmk revision fc511e40cc1a274473a753c959f8d7e5fcc317d0)
```
commit 726a482165964d67f1f8d4cd1dd3dcde67e4de57 (HEAD -> nrf_suspend)
Author: Frederik M.J.V <freqmod@gmail.com>
Date:   Sun Oct 2 12:05:30 2022 +0200

    Support for supend for nrf52

diff --git a/app/dts/behaviors/reset.dtsi b/app/dts/behaviors/reset.dtsi
index cb246814..f1192023 100644
--- a/app/dts/behaviors/reset.dtsi
+++ b/app/dts/behaviors/reset.dtsi
@@ -20,5 +20,11 @@
                        type = <RST_UF2>;
                        #binding-cells = <0>;
                };
+               suspend: behavior_reset_suspend {
+                       compatible = "zmk,behavior-reset";
+                       label = "SUSPEND";
+                       type = <RST_SUSPEND>;
+                       #binding-cells = <0>;
+               };
        };
 };
diff --git a/app/include/dt-bindings/zmk/reset.h b/app/include/dt-bindings/zmk/reset.h
index 2b3d8760..4ebb8d0f 100644
--- a/app/include/dt-bindings/zmk/reset.h
+++ b/app/include/dt-bindings/zmk/reset.h
@@ -10,4 +10,5 @@
 // AdaFruit nrf52 Bootloader Specific. See
 // https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/d6b28e66053eea467166f44875e3c7ec741cb471/src/main.c#L107

-#define RST_UF2 0x57
\ No newline at end of file
+#define RST_UF2 0x57
+#define RST_SUSPEND 0xFE
\ No newline at end of file
```
And for zephyr (on revision 8adeab429a2480198d22df9847933cfe3f9ea410 ):
```
commit d1eee565d5c3aab021839aa77743a48b32575815 (HEAD -> nrf52_suspend)
Author: Frederik M.J.V <freqmod@gmail.com>
Date:   Sun Oct 2 12:09:46 2022 +0200

    Allow soft power off of nrf52 boards

diff --git a/soc/arm/nordic_nrf/nrf52/soc.c b/soc/arm/nordic_nrf/nrf52/soc.c
index e1e63e9f8..51e4665bb 100644
--- a/soc/arm/nordic_nrf/nrf52/soc.c
+++ b/soc/arm/nordic_nrf/nrf52/soc.c
@@ -18,6 +18,7 @@
 #include <hal/nrf_power.h>
 #include <soc/nrfx_coredep.h>
 #include <logging/log.h>
+#include <pm/state.h>

 #ifdef CONFIG_RUNTIME_NMI
 extern void z_arm_nmi_init(void);
@@ -51,8 +52,13 @@ LOG_MODULE_REGISTER(soc);
    Set general purpose retention register and reboot */
 void sys_arch_reboot(int type)
 {
-       nrf_power_gpregret_set(NRF_POWER, (uint8_t)type);
-       NVIC_SystemReset();
+       if (type == 0xFE){
+               nrf_power_system_off(NRF_POWER);
+       }
+       else{
+               nrf_power_gpregret_set(NRF_POWER, (uint8_t)type);
+               NVIC_SystemReset();
+       }
 }

 static int nordicsemi_nrf52_init(const struct device *arg)
```

manna-harbour, 2022-10-09:
@freqmod To make your changes easier to test, here they are in separate branches, including a change to the zmk manifest to use the zephyr fork.
- https://github.com/manna-harbour/zmk/tree/suspend-test
- https://github.com/manna-harbour/zephyr/tree/v3.0.0+zmk-fixes-suspend-test

Support is also included in [Miryoku ZMK](https://github.com/manna-harbour/miryoku_zmk#suspend).

manna-harbour, 2022-10-09:
@freqmod Could this be implemented using `BEHAVIOR_LOCALITY_GLOBAL`?

freqmod, 2022-10-09:
Initially i tried to do global behaviour. As far as i could see from the implementation the global behaviour first sends a command to the peripheral (right side) and then executes the command itself. I was a bit concerned that the command to the peripheral side seems to be put in a queue. If the right side executes before the queue element is processed then it will power down the right side before the left side gets the command to power of. I guess there are ways to handle this e.g. by putting the command to power down the left side in the queue after the other command. Another option may be to sleep, but if you block the queue processing it does not help. I just wanted something that worked within a reasonable implementation time, so i haven't investigated this, and rather went with the local behaviour.

Btw i have been testing this for a week without any issues (except sometimes reconnecting after wakeup, but i guess that is present if you use a hard power off switch too). It seems my battery stays charged too.

jacobsamlarose, 2022-10-09:
@freqmod Question (and please excuse my noobiness): I'm testing this on a Corne with OLED displays; I usually keep the OLEDs off unless diagnosing a fault, but kept them on to test this soft shutdown. I have a "travel layer" with a keymap full of &nones that I'd usually toggle on before packing the board; I've now added &suspend keys to each side in that layer. When I press the &suspend keys, the Bluetooth connection shuts down and the keys are non-responsive (as I'd expect for a shutdown) but the OLEDs continue to display whatever was showing when I pressed the suspend keys— is that expected?

frederik-vestre-pexip, 2022-10-10:
I am not sure what the nrf52 suspend does. Have you considered seeing if the external power off behaviour turns off the displays, maybe they can be combined with this in a macro or something?

jacobsamlarose, 2022-10-10:
Something like: 
`ZMK_MACRO(shutdown,
        wait-ms = <30>;
        tap-ms = <40>;
        bindings = <&ext_power EP_OFF &suspend>;
    )`

? I'll give this a go and see.

MartianH, 2022-11-15:
Any update on this?

caksoylar, 2023-12-11:
Would be resolved by #1942.

elpekenin, 2024-04-09:
Closed by #2085

closed 2024-04-09 by caksoylar