zmkfirmware/zmk #1299: Feature Request: New &df layer behaviour

manna-harbour, 2022-05-13:
I'd like to be able to lock a momentary layer. That is, while holding `&mo` or `&lt`, tap the layer lock key then release the layer hold but stay on the layer.

`&to` can not be used as the layer is deactivated when the layer hold is released.  `&tog` can not be used as it deactivates the layer immediately.

QMK has [`DF`](https://github.com/qmk/qmk_firmware/blob/master/docs/keycodes.md#layer-switching-idlayer-switching), and KMonad has [`layer-switch`](https://github.com/kmonad/kmonad/blob/c43a0180a3b9ca9c4ff4a283b17369fc4e8e46e2/keymap/tutorial.kbd#L583), that can be used for this purpose.

I propose adding a new `&df` layer behaviour to change the default layer.

See the discord comment thread starting [here](https://discord.com/channels/719497620560543766/719565084208398406/907869038091313182) for an earlier discussion on this topic.

dabell-cc, 2022-08-03:
Could you make use of the example behaviour shown in the "Toggle-on-Tap, Momentary-on-Hold Layers" example found on the [Hold-Tap Behaviour](https://zmk.dev/docs/behaviors/hold-tap#example-use-cases:~:text=Toggle%2Don%2DTap%2C%20Momentary%2Don%2DHold%20Layers) page?

I set my keymap up to allow a 'Fn' key that will toggle the F1..F12 keys layer on when tapped, but allow Fn+F1..F12 to be used without then having to un-toggle the layer. A note of caution though, the documentation currently has the <&tog>, <&mo> bindings reversed so you'll want to switch them. (I put in a PR for it earlier today).

I suppose you might not want to dedicate a key to this behaviour, such as the `&lt` use case, where the tap might be a character entry key.

bmijanovich, 2023-08-07:
Below is an example of how to achieve a layer lock using [Conditional Layers](https://zmk.dev/docs/features/conditional-layers). By allowing two separate layers to each control the state of the layer you want to lock, you can avoid the layer deactivating when you release `&mo` after pressing `&tog` (or `&to`).

The key points here are:
* You are not limited to a single Conditional Layer condition
* Conditional Layer conditions do not require 2 or more `if-layers`; 1 will do just fine
* Since the `LOWER` layer is controlled by Conditional Layers, you cannot activate it directly; you must instead activate it indirectly using one of the `if-layers`

```
#define DEFAULT 0
#define LOWER 1
#define LOWER_TEMP 2
#define LOWER_LOCK 3

/ {
    conditional_layers {
        compatible = "zmk,conditional-layers";
        lower_temp {
            if-layers = <LOWER_TEMP>;
            then-layer = <LOWER>;
        };
        lower_lock {
            if-layers = <LOWER_LOCK>;
            then-layer = <LOWER>;
        };
    };
    keymap {
        compatible = "zmk,keymap";
        default_layer {
            bindings = <
                &mo LOWER_TEMP    &kp A    &kp B
            >;
        };
        lower_layer {
            bindings = <
                &trans    &tog LOWER_LOCK    &kp C
            >;
        };
        lower_temp_layer {
            bindings = <
                &trans    &trans    &trans
            >;
        };
        lower_lock_layer {
            bindings = <
                &trans    &trans    &trans
            >;
        };
    };
};
```

I wrote this on my phone so no guarantees about it being bug free or even compiling. The approach works though; I use it in my keymap for exactly this feature.

ecstatic-morse, 2023-11-02:
I implemented a basic version of this in #1984 and am using it locally. I'm curious how my version compares to what others had in mind. For example, I think a caps word-like option to disable locked layers would be useful, but it would be a bit difficult to configure...

txmrv, 2024-10-31:
You can use a macro for this.

```keymap
macros {
        m_lh: macro_layer_hold_on_release {
            compatible = "zmk,behavior-macro-one-param";
            #binding-cells = <1>;
            wait-ms = <40>;
            tap-ms = <40>;
            bindings =
                <&macro_pause_for_release>,
                <&macro_param_1to1>,
                <&macro_tap &to MACRO_PLACEHOLDER>;
        };
    };
```

Press the `&mo LAYER` press `&m_lh LAYER` release `&mo` release `&m_lh`

maxesser1776, 2025-03-09:
I have a question i know this is old but i was wondering if there is a way to do this with a &lt instead of a &mo

bmijanovich, 2025-03-09:
@maxesser1776 I feel like I've been summoned from the past! 🤣 I haven't touched ZMK in ages so I'm not going to try to give you sample code, but yeah you absolutely can and my keymap works this way. If you're using the example I have above, just have &lt activate your temp layer on hold.

Nick-Munnich, 2025-03-09:
While not *exactly* what is requested, this feature should be resolved by #2717

maxesser1776, 2025-03-09:
@bmijanovich thanks for the response I'm on a chocofi 3x5 i don't have room for momentary layer keys so I'm using layer tap and i have a layer that is my numpad on the right and left is setup to have keyboard shortcuts for my CAD software and i want to lock that layer on when I'm working in that software. If that makes sense. ill try and use what you have and see if i can get it to work.

bmijanovich, 2025-03-09:
@maxesser1776 I have a pretty in depth readme for my ZMK config [here](https://github.com/bmijanovich/zmk-config/blob/main/README.md). What you're trying to do sounds pretty much exactly like what I did. Fair warning though, I use a bunch of C macros to avoid repeating ZMK config code a lot, so you'll have to do some macro expansion to get a clean example.

hitsmaxft, 2025-03-26:
There are DF and PDF(persistantly switch default layer) in qmk to switch default layer, which will quite useful to people like me that sharing same keyboard with  both Windows and Mac, and using different default layout ,  to get rid of  ctrl and cmd issue.

nmunnich, 2025-03-26:
Just to document my thoughts here and avoid having them hidden in random Discord conversations:

I think the concept of a "default layer" is complete nonsense. What we have is a "bottom layer" which cannot be disabled. We then also have the concept of "layers-active-on-boot", which is currently only the "bottom layer". There is no reason we cannot have multiple layers be active on boot. 

What we then would want is a behavior to save a set of active layers to a particular endpoint (be it a device on a ble profile or usb) such that the active layers swap to the saved set when switching to that endpoint. My desire is for this to work with the locked layers from  #2717, so a behavior that saves the currently locked layers to an endpoint.

closed 2025-11-14 by caksoylar