zmkfirmware/zmk #1321: Build fails due to CMake version

LomaxOnTheRun, 2022-05-28:
I'm building the exact same basic keymap layout as I successfully ran 3 months ago (i.e. the same commit) through the automated Github actions. The error log I get is:

```
Run west build -s zmk/app -b nice_nano_v2 -- -DZMK_CONFIG=${GITHUB_WORKSPACE}/config -DSHIELD=cradio_left 
-- west build: generating a build system
-- ZMK Config directory: /__w/zmk-config/zmk-config/config
-- Board: nice_nano_v2, /__w/zmk-config/zmk-config/zmk/app/boards/arm/nice_nano, cradio_left, cradio
-- ZMK Config Kconfig: /__w/zmk-config/zmk-config/config/cradio.conf
-- Using keymap file: /__w/zmk-config/zmk-config/config/cradio.keymap
-- Using keymap file: /__w/zmk-config/zmk-config/config/cradio.keymap
Including boilerplate (Zephyr base): /__w/zmk-config/zmk-config/zephyr/cmake/app/boilerplate.cmake
CMake Error at /__w/zmk-config/zmk-config/zephyr/cmake/app/boilerplate.cmake:29 (cmake_minimum_required):
  CMake 3.20.0 or higher is required.  You are running version 3.16.3
Call Stack (most recent call first):
  /__w/zmk-config/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /__w/zmk-config/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/__w/zmk-config/zmk-config/build -S/__w/zmk-config/zmk-config/zmk/app -GNinja -DBOARD=nice_nano_v2 -DZMK_CONFIG=/__w/zmk-config/zmk-config/config -DSHIELD=cradio_left
Error: Process completed with exit code 1.
```

This happens on both halves of a `cradio` / `nice_nano_v2` build, on the `Build (nice_nano_v2, cradio_left)` step (replace `_left` for `_right` for the other half). I don't see anywhere for me to define the CMake version, and I've re-downloaded the empty project and copied files over just to make sure I'm using the latest config and it's still broken.

It looks like this line is likely the culprit:

> CMake 3.20.0 or higher is required.  You are running version 3.16.3

If anyone can give me some insight on why this is happening or what I can do to fix things I'd be super appreciative.

kurtis-lew, 2022-05-28:
Last month we had the Zephyr 3.0 upgrade. Have you seen the information here yet?
https://zmk.dev/blog/2022/04/02/zephyr-3-0

LomaxOnTheRun, 2022-05-28:
That fixed it, thank you very much for such a quick solution :smile: :tada:

closed 2022-05-28 by petejohanson