zmkfirmware/zmk #1328: Issues with configuring the `continue-list` for the `caps_word` behaviour

mattdibi, 2022-06-03:
Hi there,

first and foremost thanks for this awesome project.

I recently updated my keyboard firmware and encountered an issue with the customization of the `continue-list` for the `caps_word` behaviour.

#### Configuration

I'm using the Italian layout in my OS and thus the Italian underscore character corresponds to the question mark on the US ANSI layout.

For reference: 
[image]

Therefore I added the following in my configuration:

```cpp
#include "miryoku.h"

&caps_word {
	continue-list = <SLASH QUESTION>;
};

/ {
	keymap {
```

#### Expected behaviour

For easier understanding I'll write down the example as if I was using the **US layout on the OS**.

With the above mentioned configuration I expect to type the following after activating the `caps_word` behaviour:

```
TEST/TEST?TEST
```

**Note** that I wrote the question mark by pressing `LSHFT` and `SLASH` toghether.

#### Actual behaviour

With the above mentioned configuration I actually have the following behaviour after activating the `caps_word`:

```
TEST/TEST?test
```

i.e. the `LSHFT` +  `SLASH` (which results in the "?" character) acts as if it is not belonging to the `continue-list`

#### What I tried to do

I tried the following configuration without success:

```cpp
#include "miryoku.h"

&caps_word {
	continue-list = <SLASH LS(SLASH) RS(SLASH)>;
};

/ {
	keymap {
```

#### Questions

How do I add the `LSHFT` +  `SLASH` keypresses to the continue list? Is it possible?

dxmh, 2022-06-03:
I believe this would be because `LSHIFT` is not in your continue list, therefore it exits caps word when you press it. 

Try this:

```
&caps_word {
  continue-list = <SLASH QUESTION LSHIFT>;
};
```

mattdibi, 2022-06-03:
Hi @dxmh,

it works! Thanks :)

closed 2022-06-03 by mattdibi