zmkfirmware/zmk #1331: Unable to turn on USB debugging

LomaxOnTheRun, 2022-06-04:
I'm using the [user config repo](https://zmk.dev/docs/user-setup) to set up my keyboard, and am trying to turn on the USB logging to figure out why I can't connect to my MacOS with bluetooth. I'm following the steps [here](https://zmk.dev/docs/development/usb-logging) to enable USB logging on my keyboard, but nothing I do seems to actually add the `CONFIG_ZMK_USB_LOGGING` option to the build. I have tried to add it in a couple of places:

- `zmk-config/config/cradio.conf` (the config file generated by the user config repo script)
- `zmk-config/config/boards/shields/cradio/cradio.conf` (the file specified in the debugging docs)

but I still don't see `CONFIG_ZMK_USB_LOGGING` in the Github Actions output. I can see similar entries (`CONFIG_ZMK_USB_INIT_PRIORITY=50`, `CONFIG_ZMK_BLE_INIT_PRIORITY=50`) which makes me think I'm at least looking in the right place for `CONFIG_ZMK_USB_LOGGING`.

The actual code I have in each of the files specified above is:

```bash
# Turn on logging, and set ZMK logging to debug output
# WARNING: BATTERY INTENSIVE, TURN OFF WHEN NOT DEBUGGING
CONFIG_ZMK_USB_LOGGING=y

```

RomboutKarelse, 2022-06-04:
Does your board have CDC ACM enabled? See https://zmk.dev/blog#usb-logging

LomaxOnTheRun, 2022-06-05:
It turns out I did not, thank you @RomboutKarelse 😄 I will say that the blog isn't super clear about to enable this, as it mentions:

> To ensure that USB logging properly works with custom board definitions, two sections of the `<board>.dts` file need updating.

Where the user config repo doesn't have any `.dts` files at all. So for any future explorers what I did was to edit my `cradio.keymap` file (as I'm using a "cradio" layout) to add two bits of code:

1. Just above where my keymap layout is defined:

```
/ {
	...
	// This is the code I added
	chosen {
		zephyr,console = &cdc_acm_uart;
	};

	// This is where the keymap layout is defined
	keymap {
		...
	}
	...
}
```

2. At the very bottom of the entire keymap file:

```
// Allow USB logging
&usbd {
	status = "okay";
	cdc_acm_uart: cdc_acm_uart {
		compatible = "zephyr,cdc-acm-uart";
		label = "CDC_ACM_0";
	};
};
```

The final piece of the puzzle is to add the line `CONFIG_ZMK_USB_LOGGING=y` in the `zmk-config/config/cradio.conf` file, as per my original description.

When this was all working and I pushed up the config, I could see in the `Build (nice_nano_v2, cradio_left)` Github Action job, in the `cradio_left - nice_nano_v2 Kconfig file` step, the line `CONFIG_ZMK_USB_LOGGING=y`. Note that for some reason the Github search couldn't find it, I had to open that step's output and use the browser "Find in page..." functionality to find it.

caksoylar, 2022-06-05:
Just so future readers are not confused, you should not have needed to add those snippets to your config. Logging is set up at a board level (`nice_nano_v2` in your case) and not shield level (`cradio`), and `nice_nano_v2` board [is already updated upstream with logging changes](https://github.com/zmkfirmware/zmk/blob/505c481f6a087919c976c90efae27a2baf0ec7d3/app/boards/arm/nice_nano/nice_nano.dtsi#L18).

closed 2024-08-07 by Nick-Munnich