zmkfirmware/zmk #1346: Feature request: dynamic macros

arabshapt, 2022-06-16:
Being able to record macros(similar to vim) on the fly could be useful.

QMK supports this feature: [Link ](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_dynamic_macros.md#:~:text=QMK%20supports%20temporary%20macros%20created,combined%20total%20of%20128%20keypresses.)

nickconway, 2022-06-16:
I have some starter code for this and I plan on working on it more when I get some free time. I'd love some help with it if you're familiar with the codebase!

arabshapt, 2022-06-17:
I would gladly help, but I am not familiar with the codebase and I don't yet have a keyboard with ZMK on it. I am using QMK on a usb_to_usb_converter at the moment. The new Kinesis keyboard(advantage 360 Pro with ZMK) should come out in September. A must-buy for me:)

nickconway, 2022-06-23:
There's an open PR for this now :) #1351