zmkfirmware/zmk #1347: &rgb_ug RGB_COLOR_HSB on split keyboard is not global when used within macro

ivandaho, 2022-06-16:
When using using the example code as seen in https://zmk.dev/docs/behaviors/macros#layer-activation--more for **Layer + Underglow Color**, underglow color is only set for the central side of the keyboard during macro execution

- Color for peripheral side is unaffected by the macro
- If i change underglow color directly with `&rgb_ug RGB_COLOR_HSB(86,81,85)` outside of a macro, it changes for both sides as expected

Tested on `main` branch, built via docker method for Corne/NRFMicro
Unsure if any other `&rgb_ug` behavior is affected

reproducable in https://github.com/ivandaho/zmk/tree/corne-ivan, in case you have a corne/NRFMicro
- run `./buildcorne.sh` in `app` directory
- flash firmware
- turn on LEDs by holding these buttons on the central/left side down in this order
  - leftmost button on bottom row
  - rightmost button on bottom row
  - "D" key (4th button on 2nd row)
- press either:
  - leftmost button on bottom row (central/left side)
  - middle button on bottom row (peripheral/right side)

tokazio, 2023-01-17:
Same issue here https://github.com/tokazio/zmk-config-sofle

caksoylar, 2023-01-17:
Most likely related to #1494

caksoylar, 2024-09-20:
@aDogCalledSpot You can test #2409 which should fix this issue.

caksoylar, 2024-09-23:
This should be fixed by #2409, please report if there's an issue.

closed 2024-09-23 by caksoylar