zmkfirmware/zmk #1356: Dev documentation not accurate? (--build-dir)

schadenn, 2022-06-20:
Hi 👋 
I'm currently trying to compile zmk for a draculad keyboard.
The dev documentation has this section about split keyboards: https://zmk.dev/docs/development/build-flash#building-for-split-keyboards
However when I try to run `west build -d build/left -b nice_nano -- -DSHIELD=kyria_left`, I'm getting this error:
````
ERROR: it looks like . is a build directory: did you mean --build-dir . instead?
FATAL ERROR: refusing to proceed without --force due to above error
````

I am a full time coder, so I tried a lot of things, but I can't figure out what the error is trying to tell me/what would be the solution...?
Has anyone had the same error before?

schadenn, 2022-06-20:
Thanks for the quick response. I am inside of my zmk checkouts `/app` folder though. So this is not the fix 😢

caksoylar, 2022-06-20:
After following the instructions at https://zmk.dev/docs/development/setup for Ubuntu, I can get that command line compiling fine under the `app/` folder with [your `main` branch](https://github.com/schadenn/zmk) or upstream ZMK. Note that if you switch between ZMK commits you might need to `west update` again to get the required Zephyr versions.

caksoylar, 2023-12-11:
Closing since there is no follow up.

closed 2023-12-11 by caksoylar