zmkfirmware/zmk #1363: Positional hold-tap setting does not allow for mod chording

urob, 2022-06-25:
When using the positional hold-tap feature with home-row mods, in order to "chord" mods, one has to include same-hand HRMs in the `hold-trigger-key-positions`. E.g., if one wants to chord `LSHFT` and `LCTRL` located on the left index and middle fingers, one has to include both keys in the `hold-trigger-key-positions` for ***these*** two keys.

Unfortunately, this often defeats the purpose on keyboard layouts where the most common letters are placed on the home row as well. While it doesn't hurt to enable positional hold-taps for non-home-row keys, most misfires are likely to occur on the home-row.

It would be great to add an option that allows for chording of same-hand mods. I am not sure of what would be the best set of rules to achieve this. Below are some initial thoughts based on [a discussion](https://discord.com/channels/719497620560543766/813882537436905552/984967189058691153) on discord.

In their [QMK implementation](https://github.com/manna-harbour/qmk_firmware/tree/bilateral-combinations), @manna-harbour  managed to allow for mod-chording of keys that aren't included in `hold-trigger-key-positions`. If I understand correctly, they do so by changing the mod-hold to a tap only when the next key is a tap.

One question about this approach is how it interacts with ZMK's `global-quick-tab` setting. If `global-quick-tap` takes precedence, then the 2nd mod-tap would most likely be turned into a tap (given that chording is usually fast), and hence the exemption wouldn't apply. On the other hand, if the contemplated "next-key-hold-trigger-exemption-rule" takes precedence, this would effectively disable `global-quick-tap` on the home row. 

Finding a good rule set is definitively a challenge here and may require some more complex rules. But it would be nice to have a discussion about potential solutions.

jmding8, 2022-07-26:
It sounds like you are describing a need for tap-trigger-key-positions?

urob, 2022-07-26:
> It sounds like you are describing a need for tap-trigger-key-positions?

Is that a PR? I couldn't find anything but just based on the name this sounds indeed like what I propose.

urob, 2022-08-05:
I implemented a small proof-of-concept version that fixes the issue by applying the positional-hold-tap logic only upon key-release.

Based on my limited testing it works great. Here's the testing branch if someone wants to try it out. https://github.com/urob/zmk/tree/positional-hold-tap-on-release

If there's demand, I'm happy to polish this a bit more, add a config option to decide between the two behaviors, etc, and open a PR.

urob, 2022-08-06:
Added a PR (https://github.com/zmkfirmware/zmk/pull/1423) that adds the fix as an optional config option

closed 2023-03-12 by petejohanson