zmkfirmware/zmk #1367: Stack overflow after enabling USB logging

b10102016, 2022-06-29:
Hi,
I tried to enable the USB logging function in my customize 80% keyboard using the nrf52840 chipset
After enable the USB logging function, I will hit the stack error.

I add the following config to detect the stack error.
Connect uart to the gpio p0.09 to read the log.

CONFIG_HW_STACK_PROTECTION=y
CONFIG_MPU_STACK_GUARD=y
CONFIG_ZMK_USB_LOGGING=y
CONFIG_LOG_BACKEND_UART=y
CONFIG_ZMK_LOG_LEVEL_DBG=y
CONFIG_DEBUG=y

Full configuration:
Board: [https://github.com/b10102016/zmk/tree/bluemacro840_usb_logging_stackerror/app/boards/arm/bluemacro840](https://github.com/b10102016/zmk/tree/bluemacro840_usb_logging_stackerror/app/boards/arm/bluemacro840)
Shield: [https://github.com/b10102016/zmk/tree/bluemacro840_usb_logging_stackerror/app/boards/shields/helen80](https://github.com/b10102016/zmk/tree/bluemacro840_usb_logging_stackerror/app/boards/shields/helen80)

Build command:
west build -p -b bluemacro840_v1 -- -DSHIELD=helen80

disassembly: 
[zmk.txt](https://github.com/zmkfirmware/zmk/files/9010661/zmk.txt)

Crash log:
`�*** Booting Zephyr OS build 8adeab429a24  ***
[00:00:00.635,955] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.635,955] <inf> fs_nvs: alloc wra: 0, fc0
[00:00:00.635,955] <inf> fs_nvs: data wra: 0, 40
[00:00:00.656,372] <dbg> zmk: bvd_init: AIN7 setup returned 0
[00:00:00.662,567] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.665,985] <dbg> zmk: zmk_ble_ready: ready? 0
[00:00:00.666,015] <dbg> zmk: update_advertising: advertising from 0 to 2
[00:00:00.666,046] <err> os: ***** MPU FAULT *****
[00:00:00.666,046] <err> os:   Stacking error (context area might be not valid)
[00:00:00.666,046] <err> os:   Data Access Violation
[00:00:00.666,076] <err> os:   MMFAR Address: 0x2000af30
[00:00:00.666,076] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
[00:00:00.666,107] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
[00:00:00.666,107] <err> os:  xpsr:  0x00000000
[00:00:00.666,107] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:00.666,137] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:00.666,168] <err> os: Current thread: 0x200056e0 (unknown)
[00:00:00.778,747] <err> os: Halting system`

petejohanson, 2022-10-02:
Did you try bumping the stack sizes?

lental, 2022-10-28:
I've also noticed that  CONFIG_ZMK_LOG_LEVEL_DBG causes fatal errors. it could be a stack size issue but in general  it looks like a leak. In my development  project I was able to  transmit keypresses for a few seconds before running into a stack  trace.  With a couple other other flags enabled  I was only  able to get to BLE connection before crashing.