zmkfirmware/zmk #1372: Issue waking from sleep on OS X

vaheder, 2022-07-03:
I've recently started having an issue where my keyboard will not send keys after the computer wakes from sleep. [This is my repo](https://github.com/vaheder/zmk-config), where you can have a look at the Actions for the build versions and configurations I've been using for the last few tests. I've also included the logs in an attachment to this ticket in which the keyboard is complaining about not having an active address for the current profile when keys are pressed. The logging starts after the computer wakes from sleep, then a couple keys are pressed. [This is the build Action for the logging attached.](https://github.com/vaheder/zmk-config/actions/runs/2606245154). If I reset the left side controller via the board's reset button, it's able to communicate correctly again.

Some details about the environment:
Mac Mini, OS X 12.4
Levinson/Let's Split, wired via USB-C, not currently using bluetooth to communicate with the computer

Please let me know what further information or testing I can provide.

Logging:
[20220703_zmk_wake.txt](https://github.com/zmkfirmware/zmk/files/9035456/20220703_zmk_wake.txt)

vaheder, 2022-07-11:
Quick update: after a week of testing [this version of ZMK](https://github.com/vaheder/zmk-config/pull/4), I've not experienced the same issue after waking from sleep.

dpangier, 2022-08-29:
Having the same issue on latest main, with nice!nano and the cradio shields over USB.

co-dan, 2022-09-05:
Similar issue with the latest version of ZMK using nice!nano

droghio, 2022-09-11:
I experienced a similar issue on Windows 10l. I would put the computer to sleep, wait about a second, press a key on the keyboard. Incidentally the keyboard press would wake the computer but no further input would be recognized. Resetting the MCU, un-plugging  and re-plugging USB, or issuing the `&reset` behavior would fix the functionality.

Based on a discussion with honorless & bravekarma from the Discord I also tried using the `&out TOG` behavior but that did not resume the connection.

Logging:
[20220909_zmk_win10_wake_log.txt](https://github.com/zmkfirmware/zmk/files/9541898/20220909_zmk_win10_wake_log.txt)

closed 2022-09-11 by petejohanson

petejohanson, 2022-09-11:
@vaheder can you please test again with the latest fix, and re-open if you still have the problem?

vaheder, 2022-09-12:
Will do, thanks! I’ll probably be able to install, test, and report by the
end of next week.

vaheder, 2022-09-25:
After a few days of testing, things look good! Thanks @droghio and @petejohanson!

PR #1452 (2022-09-11): fix(usb): add USB_DC_RESUME to supported states
All credit for this one goes to @xudongzheng (thanks for helping debug this!). Should fix the issue where keyboards go unresponsive after their host machine wakes from sleep due to the USB driver entering an error state. I was able to both reliably reproduce the issue before the patch goes in and confirmed it no longer occurs post patch.

The `USB_DC_RESUME` state indicates the host event has resumed the connection. Adding it to the list of valid connection states to prevent the error when waking from sleep.

Zephyr API Link:
https://docs.zephyrp [... cut]
commit 6124d254 (2022-09-11): fix(usb): add USB_DC_RESUME to supported states

All credit for this one goes to @xudongzheng (thanks for helping debug
this!). Should fix the issue where keyboards go unresponsive after their
host machine wakes from sleep due to the USB driver entering an error
state. I was able to both reliably reproduce the issue before the patch
goes in and confirmed it no longer occurs post patch.

The `USB_DC_RESUME` state indicates the host event has resumed the connection.
Adding it to the list of valid connection states to prevent the error
when waking from sleep.

Zephyr API Link:
https://docs.zephyr [... cut]