zmkfirmware/zmk #1376: ZMK_RGB_UNDERGLOW_AUTO_OFF_IDLE disregards original on/off underglow state

slashfoo, 2022-07-04:
Kconfig affected:  `CONFIG_ZMK_RGB_UNDERGLOW_AUTO_OFF_IDLE`

Steps to reproduce:

1. Turn-on/Plug keyboard
2. Turn off underglow with `&rgb_ug RGB_OFF` (if they were not on before, can turn on with `&rgb_ug RGB_ON` to then turn off)
3. Wait for keyboard to go idle
4. Wake keyboard up.

Actual: lights turn on at wake-up from idle, regardless of the state of the lights at beginning of idle.
Expected: At wake-up, the board turns on the underglow to the same state it was when idle began. So if lights were off,  they won't turn on at wake-up.

DominikHorn, 2022-09-26:
I am facing the same issue, is there any update on this?

B-urb, 2022-11-28:
I have the same issue and additionally set the CONFIG_ZMK_RGB_UNDERGLOW_ON_START to n. It still turns on after every idle.

dimosped, 2023-08-28:
same issue with these overrides:
https://github.com/dimosped/glove80-zmk-config/blob/main/config/glove80.conf

@B-urb  @DominikHorn  did you have any luck with any specific combo?

slashfoo, 2023-08-28:
I have not, I stopped using the feature when I couldn't find a workaround.

DominikHorn, 2023-08-28:
I also gave up and stopped using the feature

elpekenin, 2024-04-06:
Im case Andrew's message on the PR didnt cause a notification... I wrote some code to hopefully fix this, would be great if you guys can test it, i dont have any RGB board to do so 🥴

jradtilbrook, 2024-04-25:
I think this is fixed with #2244 (and follow-up #2284) and can be closed now.

chalukyaj, 2026-10-28:
I can confirm that this is fixed in zmk 0.4 pre-release ([commit](https://github.com/zmkfirmware/zmk/commit/4dd95d7840eb3d5a3f59bb29fbf1ab054715ef85)) by enabling both 
CONFIG_ZMK_RGB_UNDERGLOW_ON_START=y
CONFIG_ZMK_RGB_UNDERGLOW_AUTO_OFF_IDLE=y

I had these additional configs for testing 
CONFIG_ZMK_SETTINGS_SAVE_DEBOUNCE=15000
CONFIG_ZMK_SLEEP=y
CONFIG_ZMK_IDLE_TIMEOUT=20000
CONFIG_ZMK_IDLE_SLEEP_TIMEOUT=30000

The CONFIG_ZMK_RGB_UNDERGLOW_ON_START is needed along with _AUTO_OFF_IDLE, without which the kb will always start with rgb disabled

The values are set to low values and note the settings save debounce, it needs to be less then the idle timeout to ensure current state of rgb is persisted if that was the last op you performed before letting the kb idle

With rgb off, resetting and waking up from deep sleep both start with the rgb off as expected and vice versa for the the on state. I was surprised myself since there wasn't any confirmation on this issue that I've been tracking myself due to facing it in zmk 0.3 consistently, so thought I'll let everybody know.

chalukyaj, 2026-10-31:
> 16\. I can confirm that this is fixed in zmk 0.4 pre-release ([commit](https://github.com/zmkfirmware/zmk/commit/4dd95d7840eb3d5a3f59bb29fbf1ab054715ef85)) by enabling both
>      CONFIG_ZMK_RGB_UNDERGLOW_ON_START=y
>      CONFIG_ZMK_RGB_UNDERGLOW_AUTO_OFF_IDLE=y
>      I had these additional configs for testing
>      CONFIG_ZMK_SETTINGS_SAVE_DEBOUNCE=15000
>      CONFIG_ZMK_SLEEP=y
>      CONFIG_ZMK_IDLE_TIMEOUT=20000
>      CONFIG_ZMK_IDLE_SLEEP_TIMEOUT=30000
>      The CONFIG_ZMK_RGB_UNDERGLOW_ON_START is needed along with _AUTO_OFF_IDLE, without which the kb will always start with rgb disabled
>      The values are set to low values and note the settings save debounce, it needs to be less then the idle timeout to ensure current state of rgb is persisted if that was the last op you performed before letting the kb idle
>      With rgb off, resetting and waking up from deep sleep both start with the rgb off as expected and vice versa for the the on state. I was surprised myself since there wasn't any confirmation on this issue that I've been tracking myself due to facing it in zmk 0.3 consistently, so thought I'll let everybody know.

Ok, I was wrong, the issue remains unfixed, with these configs
```
CONFIG_ZMK_SETTINGS_SAVE_DEBOUNCE=15000
CONFIG_ZMK_IDLE_TIMEOUT=20000
CONFIG_ZMK_IDLE_SLEEP_TIMEOUT=45000
```
Notice the slightly larger sleep_timeout, it allows another debounced save after the `state.on` is already set to `false` (off) by `rgb_underglow_auto_state` method in [rgb_underglow.c](https://github.com/zmkfirmware/zmk/blob/main/app/src/rgb_underglow.c#L470), so the second save runs at 30s which wasn't happening earlier due to the deep sleep timeout kicking in before that could happen, so the state remained on in the flash settings. But with a value of 45seconds the second save ends up toggling the state to off and the state before is never saved to allow for a consistent state restore. 
What's worse, inside `zmk_rgb_underglow_init` on [line 276](https://github.com/zmkfirmware/zmk/blob/main/app/src/rgb_underglow.c#L276) on this line the `state.on` being used is not real, since read_cb hasn't run yet, so it basically turns on the leds the very first wake up from deep sleep or reset, but patterns don't play, rgb animations remain frozen due to the settings read happening roughly 50ms later and realizing the state is off (from the last save during idle sleep)

I was able to fix it two different ways, but both have the same change for conditional on [line 276](https://github.com/zmkfirmware/zmk/blob/main/app/src/rgb_underglow.c#L276) which is moved inside the `CONFIG_ZMK_RGB_UNDERGLOW_AUTO_OFF_USB` block on [line 272](https://github.com/zmkfirmware/zmk/blob/main/app/src/rgb_underglow.c#L272) after state.on is read from `zmk_usb_is_powered` to prevent the rgb just starting up even if the last state was off
- Approach 1: 
  - Prevent any save requests from going through when already in idle, this works, but has a disadvantage that with a larger settings debounce time than the idle timeout, then the settings read after an rgb state change can only be performed if you keep the keyboard active to prevent the idle before the save has occurred.
- Approach 2:
  - This is the more robust approach, it adds another field to the rgb state `rgb_state_before_sleeping` and populating that when entering idle and settings save ends up persisting that flag as well if it happens during an idle, thus the state.on can be properly restored after deep sleep wakeup or reset by overriding `state.on` with `state.rgb_state_before_sleeping` 
  - Here's the patch for anyone who want's a fix before this is fixed in upstream which can be applied with `patch --forward -d ./zmk -p1 <  zmk-rgb-underglow.patch` in your kb directory
```zmk-rgb-underglow.patch
diff --git a/app/src/rgb_underglow.c b/app/src/rgb_underglow.c
index 3453fb44..d7938112 100644
--- a/app/src/rgb_underglow.c
+++ b/app/src/rgb_underglow.c
@@ -58,6 +58,7 @@ struct rgb_underglow_state {
     uint8_t current_effect;
     uint16_t animation_step;
     bool on;
+    bool rgb_state_before_sleeping;
 };
 
 static const struct device *led_strip;
@@ -221,6 +222,10 @@ static int rgb_settings_set(const char *name, size_t len, settings_read_cb read_
 
         rc = read_cb(cb_arg, &state, sizeof(state));
         if (rc >= 0) {
+            // restore on state from rgb_state_before_sleeping only when it last active, wake up or reset
+            if (state.rgb_state_before_sleeping) {
+                state.on = true;
+            }
             if (state.on) {
                 k_timer_start(&underglow_tick, K_NO_WAIT, K_MSEC(50));
             }
@@ -271,11 +276,10 @@ static int zmk_rgb_underglow_init(void) {
 
 #if IS_ENABLED(CONFIG_ZMK_RGB_UNDERGLOW_AUTO_OFF_USB)
     state.on = zmk_usb_is_powered();
-#endif
-
     if (state.on) {
         k_timer_start(&underglow_tick, K_NO_WAIT, K_MSEC(50));
     }
+#endif
 
     return 0;
 }
@@ -480,6 +484,8 @@ static int rgb_underglow_auto_state(bool target_wake_state) {
     sleep_state.is_awake = target_wake_state;
 
     if (sleep_state.is_awake) {
+        // this only works if waking up from idle, otherwise this sleep_state is reset and wake up uses setttings again
+        // and if the settings save had run between idle and deep sleep then the rgb state would be off
         if (sleep_state.rgb_state_before_sleeping) {
             return zmk_rgb_underglow_on();
         } else {
@@ -487,6 +493,8 @@ static int rgb_underglow_auto_state(bool target_wake_state) {
         }
     } else {
         sleep_state.rgb_state_before_sleeping = state.on;
+        // persist rgb_state_before_sleeping in settings flash
+        state.rgb_state_before_sleeping = state.on;
         return zmk_rgb_underglow_off();
     }
 }
```