zmkfirmware/zmk #1393: New and I expect, very important hold-tap feature

Sixzero, 2022-07-22:
So I read over all the open:issues whether there is already a proposal for this, and I found none.

**Homerows are cool.**

Whynot take it one step further by providing SHIFT homerow one additional functionality?

- Basically we could hold down the SHIFT instantly on holding down our homerow key(F) and not only when some other keys are getting pressed. 

Current hold-tap solutions decide about modifiers when other key was pressed, which works really well, if you use ZMK keyboards, but not when want to use the SHIFT homerow with non-ZMK peripherals like a mouse. 

If the modifier key would get instantly pressed, it wouldn't disturb basically anything, except for GUI mod or maybe ALT mod, but other times there is no trigger usually attached for just holding down SHIFT in any softwares.

Others suggested to create a specific layer for gaming, since the problem came up with playing a game and SHIFT + mouse cordination of units is not that easy with ZMK keyboard, but on this gamer layer the problem is still not solved, since I would sometimes need the F key, and not just an &kp SHIFT key, so we basically lose a key, which is not an option for few key keyboards. 

In theory this this sounds easy to implement, since we have already other hold-tap solutions, nearly the exact solution, of the one requested. One small modification is necessary is that, we hold the mod key until there is another key is getting pressed(simultaneously). Otherwise everything work the same as it works now. 
Would love to know how easy/hard to implement this.

Tl;dr:
We need SHIFT homerow a special hold-tap, with instant mod press and not only decided on other key press.

Sixzero, 2022-07-22:
It could be not a flavor, but another label, called "zero-hold" or "instant-hold" or anything like that. It could add it this mod behavior for any of the flavors.

Sixzero, 2022-07-22:
@urob in discord proposed this macro as a solution, which is not far from what I am trying to achieve: 
<&macro_press &kp LEFT_SHIFT>,
<&macro_press &mt LEFT_SHIFT F>`, 
<&macro_pause_for_release>,
<&macro_release  &mt LEFT_SHIFT F>,
<&macro_release &kp LEFT_SHIFT> 

Also another solution is retro-tap with 0 ms tapping term. 

Going to try both, and see what works. :)

Sixzero, 2022-07-22:
The retro-tap solution seem a little too different from the non-retro-tap holdtap solutions.

The macro one seem working, but not sure, there is definitely delay before it pressing the shift, that is what I feel.

andrewjrae, 2022-07-22:
I think this could probably just be a boolean option `hold-while-undecided`.

Sixzero, 2022-07-24:
Wow, that was a fast solution, going to test it!

Sixzero, 2022-07-24:
hold-while-undecided that it won't release the SHIFT sometimes. 
Probably, it releases mod key on the tap press only? so if the shift was triggered then it is not getting released?

Sixzero, 2022-07-24:
I use it with balanced flavor and taping-term-ms 200 exceded, so I press the behaviour button for more then 200ms, also not having retro-tap. 

Now I test 2 cases:
- If retrotap is enabled.
- If I use amcro to release the SHIFT on press end.

Sixzero, 2022-08-01:
I think it is fully done, in Niconway's dev branch, so I close the issue, until it gets its first review.

closed 2022-08-01 by Sixzero

theol0403, 2023-05-23:
@Sixzero can you give https://github.com/zmkfirmware/zmk/pull/1811 a try?