zmkfirmware/zmk #1395: Combo triggered unexpectedly

jmding8, 2022-07-23:
I have a keymap with a combo on keys `<4, 16>` which produces `&kp TAB`. This is the only way to emit TAB in this keymap.

https://github.com/jmding8/zmk/blob/mercury_3x5%2B3/app/boards/shields/mercury/mercury.keymap#L176-L180

Somehow, when I press keys 14, 15 and 16 in a combo-like way (not sure of exact press order, all keys held down together for ~1s) it sometimes triggers `&kp TAB`. This happens maybe one in six attempts? Truncated logs below, should be only the relevant bit. Full logs attached

```
...
[00:04:02.913,848] <dbg> zmk: on_hold_tap_binding_released: 16 cleaning up hold-tap
[00:04:04.462,158] <dbg> zmk: kscan_matrix_read: Sending event at 1,3 state on
[00:04:04.462,280] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 3, position: 15, pressed: true
[00:04:04.462,310] <dbg> zmk: position_state_changed_listener: 15 bubble (no undecided hold_tap active)
[00:04:04.462,371] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 15, binding name: D_LAYER_TAP
[00:04:04.462,402] <dbg> zmk: on_hold_tap_binding_pressed: 15 new undecided hold_tap
[00:04:04.463,195] <dbg> zmk: kscan_matrix_read: Sending event at 1,4 state on
[00:04:04.463,287] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 4, position: 16, pressed: true
[00:04:04.463,348] <dbg> zmk: position_state_changed_listener: 15 capturing 16 down event
[00:04:04.463,378] <dbg> zmk: decide_hold_tap: 15 decided hold-interrupt (tap-unless-interrupted decision moment other-key-down)
[00:04:04.463,409] <dbg> zmk: mo_keymap_binding_pressed: position 15 layer 8
[00:04:04.463,409] <dbg> zmk: set_layer_state: layer_changed: layer 8 state 1
[00:04:04.463,470] <dbg> zmk: release_captured_events: Releasing key position event for position 16 pressed
[00:04:04.463,500] <dbg> zmk: position_state_changed_listener: 16 bubble (no undecided hold_tap active)
[00:04:04.463,531] <dbg> zmk: position_state_down: combo: capturing position event 16
[00:04:04.463,531] <dbg> zmk: zmk_event_manager_handle_from: Listener captured the event
[00:04:04.463,562] <dbg> zmk: zmk_event_manager_handle_from: Listener captured the event
[00:04:04.464,202] <dbg> zmk: kscan_matrix_read: Sending event at 1,2 state on
[00:04:04.464,324] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 2, position: 14, pressed: true
[00:04:04.464,355] <dbg> zmk: position_state_changed_listener: 14 bubble (no undecided hold_tap active)
[00:04:04.464,385] <dbg> zmk: position_state_down: combo: capturing position event 14
[00:04:04.464,385] <dbg> zmk: release_pressed_keys: combo: releasing position event 16
[00:04:04.464,416] <dbg> zmk: position_state_down: combo: capturing position event 16
[00:04:04.464,447] <dbg> zmk: zmk_event_manager_handle_from: Listener captured the event
[00:04:04.464,447] <dbg> zmk: release_pressed_keys: combo: reraising position event 14
[00:04:04.464,477] <dbg> zmk: position_state_changed_listener: 14 bubble (no undecided hold_tap active)
[00:04:04.464,508] <dbg> zmk: position_state_down: combo: capturing position event 14
[00:04:04.464,538] <dbg> zmk: on_keymap_binding_pressed: position 46 keycode 0x7002B
[00:04:04.464,569] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x2B implicit_mods 0x00 explicit_mods 0x00
[00:04:04.464,569] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:04:04.464,599] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:04:04.464,721] <dbg> zmk: zmk_event_manager_handle_from: Listener captured the event
[00:04:04.464,752] <dbg> zmk: zmk_event_manager_handle_from: Listener captured the event
[00:04:04.580,169] <dbg> zmk: kscan_matrix_read: Sending event at 1,2 state off
[00:04:04.580,261] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 2, position: 14, pressed: false
[00:04:04.580,291] <dbg> zmk: position_state_changed_listener: 14 bubble (no undecided hold_tap active)
[00:04:04.580,322] <dbg> zmk: on_keymap_binding_released: position 46 keycode 0x7002B
[00:04:04.580,352] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x2B implicit_mods 0x00 explicit_mods 0x00
[00:04:04.580,383] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:04:04.580,383] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:04:04.580,444] <dbg> zmk: zmk_event_manager_handle_from: Listener handled the event
[00:04:04.584,197] <dbg> zmk: kscan_matrix_read: Sending event at 1,4 state off
[00:04:04.584,289] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 4, position: 16, pressed: false
[00:04:04.584,320] <dbg> zmk: position_state_changed_listener: 16 bubble (no undecided hold_tap active)
[00:04:04.584,350] <dbg> zmk: zmk_event_manager_handle_from: Listener handled the event
[00:04:04.587,158] <dbg> zmk: kscan_matrix_read: Sending event at 1,3 state off
[00:04:04.587,310] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 3, position: 15, pressed: false
[00:04:04.587,341] <dbg> zmk: position_state_changed_listener: 15 bubble (no undecided hold_tap active)
[00:04:04.587,371] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 15, binding name: D_LAYER_TAP
[00:04:04.587,432] <dbg> zmk: mo_keymap_binding_released: position 15 layer 8
[00:04:04.587,463] <dbg> zmk: set_layer_state: layer_changed: layer 8 state 0
[00:04:04.587,493] <dbg> zmk: on_hold_tap_binding_released: 15 cleaning up hold-tap

```

[keyboard log 2.txt](https://github.com/zmkfirmware/zmk/files/9173009/keyboard.log.2.txt)

jmding8, 2022-07-23:
This even happens sometimes on the very first keypresses after resetting the board. I.e. reset the board, press 14/15/16 in a combo-like fashion, get a `&kp TAB`

jmding8, 2022-07-23:
managed to get a side by side where it tabbed once, and did not tab the second time. Did a comparison of the logs: 
[image]

There are really no differences in the logs, just the key-up order (but the tab gets emitted before the key up even happens, so the key up order is irrelevant)
[tab.txt](https://github.com/zmkfirmware/zmk/files/9173030/tab.txt)
[noTab.txt](https://github.com/zmkfirmware/zmk/files/9173031/noTab.txt)

jmding8, 2022-07-23:
The combo that is acting up is distinct from my other combos in that it is one column only. I think there might be some matrix-scan weirdness going on? maybe in combination with macros or hold/tap?

jmding8, 2022-07-23:
After more testing, I was able to reproduce this issue with keys that are NOT all on the same column. This is my new keymap, with a combo in positions `<16, 17>` that emits ESC:

https://github.com/jmding8/zmk/blob/3cc38d4eb61fced6c0287cc0dfc920c54ed6d2c2/app/boards/shields/mercury/mercury.keymap#L173-L177

This combo fires sometimes when I type (15d, 16d simultaneously), 19d rapidly thereafter, (15u, 16u, 19u simultaneously). I even set `CONFIG_ZMK_COMBO_MAX_KEYS_PER_COMBO=2` in my mercury_left.conf file thinking it might help, because this misfire only happens when I include the 19d in the sequence (a third key).

caksoylar, 2022-07-23:
I was able to simplify this down to a single hold-tap and a single combo combination.

Here is a unit test that presses `RSHFT B C` in order where the first is a hold-tap hold, but instead of outputting `BC` (capitalized) it outputs `X` which is a combo of `A` and `B` (where `A` position was never pressed):
```patch
diff --git a/app/tests/combo/combos-and-holdtaps-3/events.patterns b/app/tests/combo/combos-and-holdtaps-3/events.patterns
new file mode 100644
index 00000000..833100f6
--- /dev/null
+++ b/app/tests/combo/combos-and-holdtaps-3/events.patterns
@@ -0,0 +1 @@
+s/.*hid_listener_keycode_//p
\ No newline at end of file
diff --git a/app/tests/combo/combos-and-holdtaps-3/keycode_events.snapshot b/app/tests/combo/combos-and-holdtaps-3/keycode_events.snapshot
new file mode 100644
index 00000000..843832dd
--- /dev/null
+++ b/app/tests/combo/combos-and-holdtaps-3/keycode_events.snapshot
@@ -0,0 +1,5 @@
+pressed: usage_page 0x07 keycode 0xE5 implicit_mods 0x00 explicit_mods 0x00
+pressed: usage_page 0x07 keycode 0x05 implicit_mods 0x00 explicit_mods 0x00
+pressed: usage_page 0x07 keycode 0x06 implicit_mods 0x00 explicit_mods 0x00
+released: usage_page 0x07 keycode 0x05 implicit_mods 0x00 explicit_mods 0x00
+released: usage_page 0x07 keycode 0x06 implicit_mods 0x00 explicit_mods 0x00
diff --git a/app/tests/combo/combos-and-holdtaps-3/native_posix_64.keymap b/app/tests/combo/combos-and-holdtaps-3/native_posix_64.keymap
new file mode 100644
index 00000000..2d926d59
--- /dev/null
+++ b/app/tests/combo/combos-and-holdtaps-3/native_posix_64.keymap
@@ -0,0 +1,47 @@
+#include <dt-bindings/zmk/keys.h>
+#include <behaviors.dtsi>
+#include <dt-bindings/zmk/kscan-mock.h>
+
+&mt {
+	flavor = "hold-preferred";
+};
+
+/ {
+	combos {
+		compatible = "zmk,combos";
+		combo_one {
+			timeout-ms = <40>;
+			key-positions = <0 1>;
+			bindings = <&kp X>;
+		};
+	};
+
+	keymap {
+		compatible = "zmk,keymap";
+		label = "Default keymap";
+
+		default_layer {
+			bindings = <
+			&kp A         &kp B
+			&mt RSHFT RET &kp C
+			>;
+		};
+	};
+};
+
+&kscan {
+	events = <
+// WORKS
+	ZMK_MOCK_PRESS(1,0,100)
+	ZMK_MOCK_PRESS(0,1,50)
+	ZMK_MOCK_PRESS(1,1,50)
+	ZMK_MOCK_RELEASE(0,1,50)
+	ZMK_MOCK_RELEASE(1,1,50)
+// FAILS
+//	ZMK_MOCK_PRESS(1,0,10)
+//	ZMK_MOCK_PRESS(0,1,10)
+//	ZMK_MOCK_PRESS(1,1,10)
+//	ZMK_MOCK_RELEASE(0,1,50)
+//	ZMK_MOCK_RELEASE(1,1,50)
+	>;
+};
```

This is the failing test output:
```diff
Running tests/combo/combos-and-holdtaps-3:
--- tests/combo/combos-and-holdtaps-3/keycode_events.snapshot   2022-07-23 13:48:39.214436488 -0700
+++ build/tests/combo/combos-and-holdtaps-3/keycode_events.log  2022-07-23 13:48:43.044436523 -0700
@@ -1,5 +1,3 @@
 pressed: usage_page 0x07 keycode 0xE5 implicit_mods 0x00 explicit_mods 0x00
-pressed: usage_page 0x07 keycode 0x05 implicit_mods 0x00 explicit_mods 0x00
-pressed: usage_page 0x07 keycode 0x06 implicit_mods 0x00 explicit_mods 0x00
-released: usage_page 0x07 keycode 0x05 implicit_mods 0x00 explicit_mods 0x00
-released: usage_page 0x07 keycode 0x06 implicit_mods 0x00 explicit_mods 0x00
+pressed: usage_page 0x07 keycode 0x1B implicit_mods 0x00 explicit_mods 0x00
+released: usage_page 0x07 keycode 0x1B implicit_mods 0x00 explicit_mods 0x00
FAILED: tests/combo/combos-and-holdtaps-3
```
(`0xE5=RSHFT`, `0x05=B`, `0x06=C`, `0x1B=X`)

jmding8, 2022-07-24:
This seems to arise when there is a hold-tap key that is also a combo key. I was able to work around the issue for now by moving my combo off of the hold-tap key.

caksoylar, 2022-09-21:
Assuming this is also fixed by #1411 right? Please reopen if not.

closed 2022-09-21 by caksoylar