zmkfirmware/zmk #1402: Keycodes

shrekale, 2022-07-25:
I saw that not all the keycodes are exposed, not all the keycodes are added. There are 2-3 things that need to be done when you add a new keycode. But then, manually adding them is quite cumbersome. We know that the HID codes are standardized and ZMK users might need access to all of them.
So, there needs to be a standard solution. The code should read these HID codes from the standard's PDF file and have them available or added, all of them. For some reason, adding them manually to keys.h seems too much of a redundant task. The new solution should be so that, future changes to the standard should automatically be absorbed by the code without any code changes.

joelspadin, 2022-08-01:
While I like this idea in theory, I see several problems with it:

1. How do we parse the PDF file?
2. How do we determine a code symbol for each key code? The standard document just provides descriptive text like "Keyboard Non-US # and ˜",  "Keyboard - and (underscore)", and "Keyboard Return (ENTER)" which are not necessarily valid to use in a keymap file.
3. How do we determine shifted key codes? Looking for "X and Y" in the document doesn't work, because "Keypad 1 and End" indicates the variants for numlock on/off, not shift on/off.
4. Assuming we find a solution for 2 and 3, what happens if a revision to the document changes the wording of an existing key code? How do we prevent that from breaking existing keymaps?

A more reasonable solution to avoiding dual maintenance would be to have one source file for key codes from which both keys.h and the documentation are dynamically generated.

shrekale, 2022-08-01:
> While I like this idea in theory, I see several problems with it:
>
> 1. How do we parse the PDF file?
> 2. How do we determine a code symbol for each key code? The standard document just provides [...]

Absolutely, that's what I was thinking. If we can't build the functionality to make the keymap dynamically from the PDF, we can at least write a helper script that can generate a base version of keys.h and other files for us. And then we review the keycodes manually (better than handtyping all of them). This way at least the first effort of manually typing in the codes in the file is alleviated.

closed 2022-08-06 by shrekale