zmkfirmware/zmk #1410: Feature request: continue-mod-list for caps-word

urob, 2022-07-30:
The current behavior is that caps-word continues on all mod. It would be useful to add a configuration option that allows the user to specify for which mods caps-word continues and for which it is deactivated.

Relatedly, would deactivating caps-word on mod-presses be a more sensible default?

urob, 2022-08-05:
Upon further consideration, having a complete `continue-mod-list` might be overkill. In most use cases, a simple `ignore-modifiers` property that can be turned on or off by the user should be enough. To that effect, I added a PR implementing that feature: https://github.com/zmkfirmware/zmk/pull/1422

urob, 2022-09-10:
This is fixed by https://github.com/zmkfirmware/zmk/pull/1451. With the PR one can use `/delete-property ignore-modifiers/` and then use `continue-list` to specify for which modifiers caps-word is *not* cancelled. For example, to continue caps-word on shift but cancel it with all other modifiers, one would add the following to the user-config:
```
&caps_word {
    /delete-property/ ignore-modifiers;
    continue-list = <UNDERSCORE  BACKSPACE DELETE LSHFT RSHFT>;
};
```

aadcg, 2026-09-02:
I have used this feature both in QMK and ZMK and I believe that the former has a saner default. 

@gregorias mentioned a pitfall of the current default. Another one is Alt+Tab becoming Alt+Shift+Tab. 

> (QMK docs) Caps Word automatically disables itself at the end of the word. That is, it stops by default once a space or any key other than KC_A--KC_Z, KC_0--KC_9, KC_MINS, KC_UNDS, KC_DELETE, or KC_BACKSPACE is pressed. Caps Word also disables itself if the keyboard is idle for 5 seconds.