zmkfirmware/zmk #1417: Battery drain on peripheral side of split KB using nice!nano v2

abonham, 2022-08-02:
Thanks for all the hard work on an amazing firmware!

I'm using a Ferris Sweep / Corne with a pair of nice!nano v2's and I'm seeing extreme battery drain on the peripheral side. I've tested on macOS 12 and Win10, no difference there.

cradio.conf:

caksoylar, 2022-08-03:
Just one theory, but the very high battery use on the peripheral side might be due to a short between RAW/GND pins like a flux bridge, which would continuously leak some current. For the central side it might just be due to the high idle timeout; the default value is 10 seconds and you are setting it to two minutes.

Also a heads up: You might get better and more timely help in the ZMK Discord since debugging these issues with real-time conversations is easier, and potentially has a larger audience that can help: https://zmk.dev/community/discord/invite

abonham, 2022-08-04:
Thank you so much for very usable feedback! Will go over it with a microscope and see what I can see!

abonham, 2022-08-10:
Update: it was indeed a bad nice!nano, still waiting to hear back on getting a warranty replacement, but I purchased a new one in the mean time. Issue has been resolved so far; after a full day of use, battery has only dropped 2% rather than 300%.

closed 2022-08-10 by abonham

n0x00x, 2023-06-02:
> Update: it was indeed a bad nice!nano, still waiting to hear back on getting a warranty replacement, but I purchased a new one in the mean time. Issue has been resolved so far; after a full day of use, battery has only dropped 2% rather than 300%.

How did you determine that nano was bad? i think im having the same issue :(

stiliajohny, 2023-12-05:
Following 

Same issue